espressif/esp-mqtt #157: AWS MQTT Connection issue

Schuck84, 2020-05-18:
Hello,
I have a ESP32-S2-SAOLA-1R Board and would like to connect to the AWS IoT Cloud Services.
Currently I'm using the example from esp-idf\examples\protocols\mqtt\ssl_mutual_auth
I have copied the .cert.pem and .private.key to the projects main folder where the app_main.c file is located.
I renamed the files as followed:
.cert.pem -> client.crt
.private.key -> client.key

Additionally I entered my WiFi Credentials in the sdkconfig and I'm able to connect to the internet.

After compilation and flashing the program I get the following messages in the terminal:

I (4553) example_connect: Connected to C3PO
I (4553) example_connect: IPv4 address: 192.168.0.39
I (4563) example_connect: IPv6 address: fe80:0000:0000:0000:7edf:a1ff:fe00:76d6
I (4573) MQTTS_EXAMPLE: [APP] Free memory: 135716 bytes
I (4583) MQTTS_EXAMPLE: Other event id:7
E (8543) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (8543) esp-tls: Failed to open new connection
E (8543) TRANS_SSL: Failed to open a new connection
E (8553) MQTT_CLIENT: Error transport connect
I (8553) MQTTS_EXAMPLE: MQTT_EVENT_ERROR
I (8563) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (23563) MQTTS_EXAMPLE: Other event id:7
E (27473) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (27473) esp-tls: Failed to open new connection
E (27483) TRANS_SSL: Failed to open a new connection
E (27493) MQTT_CLIENT: Error transport connect

Could you give me some information how I can connect to the AWS cloud by using an example from this repository?
In the best case with a step by step guide or some reference to avoid any pitfalls.

david-cermak, 2020-05-18:
Hi @Schuck84 

Seems like missing a server certificate? Please note that the `ssl_mutual_auth` example actually implements client side authentication only. Please have a look at the plain [ssl example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl) and add server certificate as outlined here: https://github.com/espressif/esp-idf/blob/a3520970f63314d4e1e94b2fb23ff7a4e1496d98/examples/protocols/mqtt/ssl/main/app_main.c#L117
(Naming is not very clear here, sorry, should probably rename the example mutual -> client-side auth...)

Schuck84, 2020-05-19:
Hi,
the authentication for AWS uses Application Layer Protocol Negotiation (ALPN), which shall be normally supported by mbedTLS. Can you confirm that the implementation in the repo support this?
Actually what I have from the AWS for authentication are two pem files with public and private key and a ID.
More information can be found here:

https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html

Is this somehow supported by in the actual implementation?
Is there an example that could be used?

david-cermak, 2020-05-19:
@Schuck84 Yes, the `esp-mqtt` does support ALPN, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#_CPPv4N24esp_mqtt_client_config_t11alpn_protosE, but we do not have any sample application demonstrating connection to AWS.
You can use https://github.com/espressif/esp-aws-iot framework which includes some ready to go AWS IoT [examples](https://github.com/espressif/esp-aws-iot/tree/master/examples)

Schuck84, 2020-05-19:
Hi @david-cermak 
I checked your info, that‘s looks quite promising.
Do you know if the actual version is compatible with ESP-IDF v4.2, as this supports ESP32-S2?

Thanks :)

Schuck84, 2020-05-20:
Hi @david-cermak,
I got a connection, it was not quite simple because I also had to add some policy to the generated certificate in the IAM Console...
Unfortunately I get other errors now as mentioned here:
https://github.com/espressif/esp-aws-iot/issues/13#issuecomment-631282275

Many thanks for your support and the hint to the repo :)
Maybe you also have an idea about the other issue?

david-cermak, 2020-05-20:
@Schuck84 Good to hear that the connection worked. I'm not familiar with aws-iot, sorry. Would suggest though translating the backtrace using the addr2line utility
```
xtensa-esp32-elf-addr2line -pfiaC -e build/PROJECT.elf ADDRESS
```
for the maintainers of https://github.com/espressif/esp-aws-iot to see where the crash originated. (see [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#automatic-address-decoding) for details)

would you mind closing the issue if no problem in this repo?

closed 2020-05-20 by Schuck84