espressif/esp-mqtt #160: ADVICE on Application MQTT with esp idf (network sensor)

SERIDJ, 2020-06-26:
hi everyone,

I'd like to ask if anyone can help me figure out the "best practice" architecture for my relatively simple ESP32 application please. I'm using the ESP-IDF with C

i want to devlop an network sensor (max 8 at the same time ) with i2cdev library , and send and receive data with mqtts to raspberry broker (mosquitto) (using queue , mutex, extern variable , timer ....)

what the difference btw this library and those https://github.com/espressif/esp-idf/tr ... utual_auth

If anyone can offer me some advice on how to go about structuring this kind of application I'd really appreciate it.

thanks

ESP-Marius, 2020-06-29:
I suggest you start with [mqtt examples](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt) to get an overview how to build a simple MQTT application using ESP-IDF.

There is no difference between this library and those https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt. Those examples are a part of the ESP-IDF repository, which uses this library as its MQTT library (it's added as a submodule [here](https://github.com/espressif/esp-idf/tree/master/components/mqtt)).

For some general design ideas you can take a look at [AWS MQTT guide](https://d1.awsstatic.com/whitepapers/Designing_MQTT_Topics_for_AWS_IoT_Core.pdf), some of it is AWS specific, but it also contains some best practices regarding topics etc.

Also, we generally prefer to keep these kind of "how to" questions on our forums: https://www.esp32.com/
That way the whole community can contribute with their comments, and its "workflow" is more suited for sharing this kind of knowledge. :smile:

david-cermak, 2020-08-20:
Closing as the question is answered.
@SERIDJ Please use esp-forum for such general questions. Thanks!

closed 2020-08-20 by david-cermak