espressif/esp-mqtt #164: Reconnect to MQTT Broker fails

keerthimk, 2020-07-24:
Hello People,

Am trying to do the following with ESP8266. I use ESP8266 FreeRTOS SDK.

1. Connect Wifi  -Initialisation
2. Start MQTT Client -Initialisation
3.Measure the Sensor data and when there is a significant change in the Sensor Data (DHT 22 Sensor), **Connect + Start MQTT Client and Publish** or **Only Publish if MQTT Client is already Connected**,
5. Disonnect Wifi
6. Stop MQTT Client

**Problem Description:**
Resets are thrown upon Reconnect
PS: Disconnects gracefully without any errors. 

Please let me know if I have a wrong sequence for reconnecting and disconnecting from MQTT Broker.
I have attached the code for your reference.

Please find the details below 
```
[query.zip](https://github.com/espressif/esp-mqtt/files/4972120/query.zip)

I (22514) MQTT_Publish_Task: Result of nvs_flash_init : 0
I (22516) MQTT_Publish_Task: Result of esp_event_loop_create_default : 0
I (22523) MQTT_Publish_Task: Result of example_set_connection_info : 0
I (22547) example_connect: Connecting to vC...
I (22551) example_connect: Connected to vC
I (22553) example_connect: IPv4 address: 192.168.9.141
I (22559) MQTT_Publish_Task: Result of example_connect : 0
I (22568) MQTT_APP_START: Client has Re-started
I (22578) TRANS_TCP: Trying to tcp_connect...
tcp_connect to port 1883
I (25066) wifi: state: 0 -> 2 (b0)
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
I (25073) wifi: state: 2 -> 3 (0)
I (25081) wifi: state: 3 -> 5 (10)
I (25084) wifi: pm start, type: 1
tcp_pcb_purge
tcp_pcb_purge: not all data sent
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
I (25582) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40101469  PS      : 0x00000033  A0      : 0x401013f7  A1      : 0x3ffed7b0
0x40101469: _heap_caps_malloc at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/heap/src/e
sp_heap_caps.c:169

0x401013f7: ptr2memblk_size at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/heap/port/es
p8266/include/priv/esp_heap_caps_priv.h:132
 (inlined by) _heap_caps_malloc at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/heap/src
/esp_heap_caps.c:125

A2      : 0x3fff3c20  A3      : 0x00000004  A4      : 0xc000c3e4  A5      : 0x3fff423c
A6      : 0x3fff3c20  A7      : 0x00000004  A8      : 0x00000625  A9      : 0x0000ffff
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x00000001  A13     : 0x00000004
A14     : 0x3fff3c28  A15     : 0x00000614  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c

Backtrace: 0x40101469:0x3ffed7b0 0x402253b4:0x3ffed7d0 0x4022322f:0x3ffed7e0 0x40228bcf:0x3ffed810 0x4022927c:0
x3ffed820 0x40232e75:0x3ffed890 0x40233984:0x3ffed8c0 0x40228730:0x3ffed8d0 0x402287e4:0x3ffed8e0
0x40101469: _heap_caps_malloc at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/heap/src/e
sp_heap_caps.c:169

0x402253b4: mem_malloc at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/cor
e/mem.c:238

0x4022322f: pbuf_alloc at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/cor
e/pbuf.c:1387

0x40228bcf: tcp_pbuf_prealloc at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip/lwip/
src/core/tcp_out.c:264

0x4022927c: tcp_write at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core
/tcp_out.c:610

0x40232e75: lwip_netconn_do_writemore at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lw
ip/lwip/src/api/api_msg.c:2173

0x40233984: lwip_netconn_do_write at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip/l
wip/src/api/api_msg.c:2173

0x40228730: tcpip_thread_handle_msg at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip
/lwip/src/api/tcpip.c:596

0x402287e4: tcpip_thread at C:/msys32/home/keerthi.muthukrishna/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/a
pi/tcpip.c:596
```

[query.zip](https://github.com/espressif/esp-mqtt/files/4972123/query.zip)

[query.zip](https://github.com/espressif/esp-mqtt/files/4972122/query.zip)

ESP-Marius, 2020-08-06:
Hi @keerthimk 

It looks like you are trying to stop and delete the MQTT client from the event handler, which unfortunately wont work. 

I suggest you try to notify another task (using a queue, event group etc) and then shut it down from there.

keerthimk, 2020-08-14:
Hi @ESP-Marius 
Thanks for your Comment.
It works when I resume another task from the event handler. 
Also I increased the Stack size of the task to 2048.

closed 2020-08-14 by keerthimk