espressif/esp-mqtt #168: QOS Level 2 Duplicate Message

jesse-schein, 2020-08-28:
I'm using the component in an Arduino based project and notice that after sending messages with QOS 2, then turning off my MQTT server, and then turning back on server, the ESP32 is resending the messages that were sent successfully, the only difference instead of high message ids, it's resetting them back to 1. I verified what its sending via wireshark.

Both have a value of 10456 which is the uptime tick count from millis().

```
const esp_mqtt_client_config_t mqtt_cfg = {
        .event_handle = mqtt_event_handler, 
        .host = host, 
        .uri = url,
        .port = port,
        .client_id = NULL,
        .username = NULL,
        .password = NULL,
        .lwt_topic = NULL,
        .lwt_msg = NULL,
        .lwt_qos = DEFAULT,
        .lwt_retain = NULL,
        .lwt_msg_len= DEFAULT,
        .disable_clean_session = 0,
        .keepalive = 10
    };
```

`   esp_mqtt_client_publish(client, "test/hello", String(millis()).c_str(), 0, 2, 0);`

Message 58872:
[image]
[image]

 
 
 
 
Message 1 (actually sends while server is shutting down):
[image]

ESP-Marius, 2020-09-01:
Hi @jesse-schein, thanks for reporting this.

Which commit of esp-mqtt are you using? The latest one?

jesse-schein, 2020-09-01:
> Hi @jesse-schein, thanks for reporting this.
> 
> Which commit of esp-mqtt are you using? The latest one?

Yes I'm using the latest master branch. I may try a different broker to see if that matters but based on what I'm seeing my broker is doing the proper MQTT flow in wireshark.

I'm a bit new to MQTT at this low level but from what I've studied up on in the event a reconnect, it should stop sending the enqueued messages if the first attempt is successful.

ESP-Marius, 2020-09-04:
Yes, if the message was confirmed the MQTT client shouldn't resend it.

But I notice in your wireshark log that it seems like the msg id = 1 resent message is actually coming from the server, and not the esp. Is this correct?

Tried to reproduce this, but wasn't able to. Which MQTT server are you using?

jesse-schein, 2020-09-04:
> Yes, if the message was confirmed the MQTT client shouldn't resend it.
>
> But I notice in your wireshark log that it seems like the msg id = 1 resent message is actually coming from the server, and [...]

Ahhh you know what, I think I sent the message from my ESP to a topic that I was subscribed too on my ESP as well. Thus the message gets sent from ESP <> Server, then Server <> ESP. Thanks for pointing out my error.

I am using C# MQTT broker. This one:  [https://github.com/chkr1011/MQTTnet](https://github.com/chkr1011/MQTTnet)

You can close this issue as resolved.

closed 2020-09-04 by jesse-schein