espressif/esp-mqtt #171: How to Link ali-mqtt

hzhh110, 2020-10-12:
How can I directly use this framework to connect to ali-mqtt? Can you give me a sample? All online are connected with linkiot framework and arduino

david-cermak, 2020-10-23:
Hi @hzhh110 

Unfortunately, we have no example of connecting to ali-mqtt. You can follow the steps described here https://partners-intl.aliyun.com/help/doc-detail/73742.htm to initiate a connection

For calculating hmac-sha1, you can use mbedtls API, for example:
``` C
    unsigned char digest[20];
    mbedtls_md_context_t md_ctx;
    const mbedtls_md_info_t *md_info = mbedtls_md_info_from_type( MBEDTLS_MD_SHA1 );

    mbedtls_md_setup( &md_ctx, md_info, 1 );
    mbedtls_md_hmac_starts( &md_ctx, key, sizeof(key) );
    mbedtls_md_hmac_update( &md_ctx, buffer, sizeof(buffer) -1);
    mbedtls_md_hmac_finish( &md_ctx, digest );
```

And init the mqtt client (using the client data from the example in https://partners-intl.aliyun.com/help/doc-detail/73742.htm)
``` c
    esp_mqtt_client_config_t mqtt_config = {
             ...

            .client_id = "12345|securemode=3,signmethod=hmacsha1,timestamp=789|",
            .username = "device&pk",
            .password = ... // ==digest calculated as above, where 
//    const unsigned char buffer[] = "clientId12345deviceNamedeviceproductKeypktimestamp789";
//    const unsigned char key[] = "secret";
           ...
    };

```

david-cermak, 2021-01-22:
@hzhh110 Did the suggested calculation work?
Closing now, as it's not an issue of the `esp-mqtt`.

closed 2021-01-22 by david-cermak