espressif/esp-mqtt #172: esp_mqtt_client_publish Semaphore lock if the wifi/network is not connected yet

umer-ilyas, 2020-10-27:
steps to reproduce issue

- wifi initialized but not connected to AP yet
- mqtt client started `esp_mqtt_client_start`
- if i call `esp_mqtt_client_publish`  then     `MQTT_API_LOCK(client);` does not return 

> I have not figured out where is sempore is acquired but it is something to do with wifi connection

- after few seconds wifi connects and  `MQTT_API_LOCK(client);` gets the lock and everything return to normal 

if for some reason wifi AP is not available `MQTT_API_LOCK(client);`  in function `esp_mqtt_client_publish` will stuck forever

david-cermak, 2020-11-02:
Hi @umer-ilyas 

Thanks for reporting. The reason why the publishing gets stuck is that the `esp_mqtt_client_start()` tries to connect to the broker in the initial phase. This should however fail in the condition you've described and unlock the mutex for the publish to continue. And as far as I can see it works like that when tried to publish just after starting the client and before connecting to the AP:
```
I (1015) MQTT_EXAMPLE: [APP] Before publish
D (1015) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=7
I (1025) MQTT_EXAMPLE: Other event id:7
E (1035) TRANS_TCP: [sock=54] connect() error: Host is unreachable
E (1045) MQTT_CLIENT: Error transport connect
D (1045) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=0
I (1055) MQTT_EXAMPLE: MQTT_EVENT_ERROR
D (1055) MQTT_CLIENT: Reconnect after 10000 ms
D (1065) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=2
I (1075) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
D (1075) MQTT_CLIENT: mqtt_enqueue id: 969, type=3 successful
D (1085) OUTBOX: ENQUEUE msgid=969, msg_type=3, len=23, size=23
D (1085) MQTT_CLIENT: Publish: client is not connected
I (1095) MQTT_EXAMPLE: [APP] After publish
```

This is what I tried to do:
``` c
    esp_mqtt_client_start(client);
    ESP_LOGI(TAG, "[APP] Before publish");
    esp_mqtt_client_publish(client, "/topic/qos1", "data_3", 0, 1, 0);
    ESP_LOGI(TAG, "[APP] After publish");
```
and it seemed to work as expected:
* client started (mutex locked when trying to reach the broker)
* publish stuck on the lock waiting for the mutex
* client connection failed unlocking the mutex
* publish done (msg moved to the outbox)

Are you seeing a different behaviour?

umer-ilyas, 2020-11-06:
@david-cermak i was expected the same behaviour as you mentioned  
but i am getting a different behaviour  
let me digg little bit deeper and get back

david-cermak, 2021-01-22:
@umer-ilyas Any update? Closing for now. Feel free to reopen once applicable.

closed 2021-01-22 by david-cermak