espressif/esp-mqtt #179: Cannot turn off mqtt keep alive

AxelLin, 2020-11-18:
The Keep Alive value in mqtt document:
A Keep Alive value of zero (0) has the effect of turning off the keep alive mechanism.
This means that, in this case, the Server is not required to disconnect the Client on the grounds of inactivity.

But in the esp-mqtt implementation:

    if (config->keepalive) {
        client->connect_info.keepalive = config->keepalive;
    }
    if (client->connect_info.keepalive == 0) {
        client->connect_info.keepalive = MQTT_KEEPALIVE_TICK;
    }

So it cannot turn off the keep alive mechanism. This seems wrong.

david-cermak, 2020-11-18:
Thanks for the report, will have to add a means to disable the keepalive.

Would this configuration help to workaround the issue for now?
```
        esp_mqtt_client_config_t mqtt_config = {
             ...
             .keepalive = 0x10000;
             ...
        };
```
(the config value is `int`, 32bit on xtensa, but the connect msg contains lower 16bits)

AxelLin, 2020-11-18:
This trick can send keepalive=0 in mqtt connect but below logic is still wrong because it will send esp_mqtt_client_ping()
esp_mqtt_task:
case MQTT_STATE_CONNECTED:
if (platform_tick_get_ms() - client->keepalive_tick > client->connect_info.keepalive * 1000 / 2) {
...

david-cermak, 2020-11-18:
of course, this is just to get around the current limitation. It would still send pings if the client keeps quiet for the next 9 hours (`keepalive = 0x10000`, or 68 years for `keepalive = 0x7FFF0000`) which might still be applicable to some use-cases.

As mentioned before, there's no way to disable the keepalive mechanism at the moment and we will have to add support for this.

closed 2020-11-19 by david-cermak