espressif/esp-mqtt #186: esp-mqtt is a disappointing library

nikhilavrio, 2021-03-08:
esp-mqtt is a disappointing library

Please suggest working example that reconnects to MQTT Server after Wifi Disconnects and does not crash the Esp32.

We are using it for IoT Device that runs 24x7.

nikhilavrio, 2021-03-08:
[image]

euripedesrocha, 2021-03-09:
Hi, sorry for your trouble. 
Can you please add more information so we can help you?
I would suggest you to add `esp_log_level_set("MQTT_CLIENT", ESP_LOG_DEBUG)` to your code, and also share with us the ESP32 crash for better help. 

After we solve your issue, we'll work to improve our examples thanks for the feedback.

ustccw, 2021-04-08:
@nikhilavrio what a nice mqtt library! Open source, simple code, resource saving, how could you say that...  
It is suggested to provide a simple example that can be reproduced.

nikhilavrio, 2021-04-12:
@ustccw 

Mqtt over SSL is not that reliable. I've seen ESP32 crash after Wifi Disconnects.

We migrated our code to Websocket (WSS) and it seems to work fine.

nikhilavrio, 2021-04-22:
Hi,

Wifi Reconnects are an issue.

Sometimes there is no internet on Wifi. 

Can you please let me know how to check reestablish mqtt connection in a while loop ?

Calling esp_mqtt_client_reconnect crashes the system.

euripedesrocha, 2021-04-22:
Hi @nikhilavrio,
could you please add the logs for the crash? 

Also, could you give us more information on the version of idf and esp-mqtt you are using?

david-cermak, 2021-05-20:
@nikhilavrio We would very much like to understand the issue you're seeing and fix it. Please note that the `esp-mqtt` library causing crashes on disconnects or refusals to connect are serious bugs which need immediate attention, but we'd need some help from your end, at least the version you use and a snippet of user application. Do you use arduino or IDF?

---
> Can you please let me know how to check and reestablish mqtt connection in a while loop ?

Any example from [`example/protocols/mqtt`](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt) should work reconnect correctly on Wifi disconnection and indeed testes that for idf versions `v3.3` up to `v4.4-dev` and getting:
```
I (7568) MQTTS_EXAMPLE: MQTT_EVENT_DATA
I (22158) wifi:bcn_timout,ap_probe_send_start
I (24658) wifi:ap_probe_send over, resett wifi status to disassoc
I (24658) wifi:state: run -> init (c800)
I (24658) wifi:pm stop, total sleep time: 12386917 us / 22430711 us

W (24658) wifi:<ba-del>idx
I (24668) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (24668) example_connect: Wi-Fi disconnected, trying to reconnect...
E (24678) TRANSPORT_BASE: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 54
E (24688) MQTT_CLIENT: Poll read error: 119, aborting connection
I (24698) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (39038) example_connect: Wi-Fi disconnected, trying to reconnect...
I (39038) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (39038) wifi:state: init -> auth (b0)
I (39048) wifi:state: auth -> assoc (0)
I (39048) wifi:state: assoc -> run (10)
I (39068) wifi:connected with DavidsAP, aid = 1, channel 6, BW20, bssid = xx:xx:xx:xx:xx:xx
I (39068) wifi:security: WPA2-PSK, phy: bgn, rssi: -33
I (39068) wifi:pm start, type: 1

W (39098) wifi:<ba-add>idx:0 (ifx:0,  xx:xx:xx:xx:xx:xx), tid:0, ssn:0, winSize:64
I (39128) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (39628) esp_netif_handlers: example_connect: sta ip: 192.168.2.28, mask: 255.255.255.0, gw: 192.168.2.1
I (39628) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.2.28
I (39698) MQTTS_EXAMPLE: Other event id:7
I (40628) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:32ae:a4ff:fec9:f198, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (41898) MQTTS_EXAMPLE: MQTT_EVENT_CONNECTED
```

Looking at your screenshot, seeing the error handles from `esp-tls` I would guess the idf could be `v4.2`? It this true? Quite recent version, where we shouldn't expect any trouble.

---
> esp-mqtt is a disappointing library

There were indeed some stability/robustness issues with this library in the past, but most of them were fixed over a year ago. I'd say that approximately from [this version](https://github.com/espressif/esp-mqtt/commits/9a5187771a7fae4a45b532a6284f63a66d9d80f7), the esp-mqtt is considered quite reliable library.

david-cermak, 2021-06-24:
@nikhilavrio Any update on this issue?

Closing it for now for lack of feedback. Please feel free to reopen with more details.

closed 2021-06-24 by david-cermak