espressif/esp-mqtt #188: Crashes on ESPIDF v4.4

calkoe, 2021-05-13:
Hello,

in combination with ESPIDF4.4 the ESP Crashes while trying to connect to the Server, 
i got the following error-message:

```
E (3191) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (3192) esp-tls-mbedtls: Failed to set client configurations
E (3197) esp-tls: create_ssl_handle failed
E (3200) esp-tls: Failed to open new connection
E (3205) TRANSPORT_BASE: Failed to open a new connection
E (3212) MQTT_CLIENT: Error transport connect
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

In ESPIDF v4.2.1 everything works fine. Unfortunately a Arduino-Framework based on ESPIDF4.2.1 is not available,
the only available v4.x-Version is v4.4.

Is there a way to get this running on v4.4 ?

Thanks for the support!

david-cermak, 2021-05-20:
@calkoe Could you please attach a snippet of your initialization? As mentioned in the related https://github.com/espressif/arduino-esp32/issues/5176 , ESP-IDF refuse to connect in an insecure way (under default config from `v4.3`), but it should not crash like that. It should just report error using `esp_event` with `MQTT_EVENT_ERROR` id and continue (which by just disconnects and retries after some time by default).

calkoe, 2021-05-20:
I'm not able to reproduce the Crash with the latest version or #2.0.0-alpha1.
I don't know if i missed specific conditions from my tests 7 days ago,
i will keep working with MQTT-TLS-Connections and reopen the issue if i manage to reproduce this issue.

closed 2021-05-20 by calkoe