espressif/esp-mqtt #189: Hostname resolution failed

nikhilavrio, 2021-05-17:
I (229757) MQTT_APP: MQTT_EVENT_ERROR
I (229762) MQTT_APP: Last error code reported from esp-tls: 0x8001
I (229768) MQTT_APP: Last tls stack error number: 0x0
I (229774) MQTT_APP: MQTT_EVENT_DISCONNECTED
I (234779) MQTT_APP: Other event id:7
E (241737) esp-tls: couldn't get hostname for :test.server.com:
E (241737) esp-tls: Failed to open new connection
E (241738) TRANS_SSL: Failed to open a new connection
E (241744) TRANSPORT_WS: Error connecting to host test.server.com:443
E (241752) MQTT_CLIENT: Error transport connect

david-cermak, 2021-05-20:
@nikhilavrio 

Yes, this error message means that the `test.server.com` hostname cannot be resolved.

Here's how you can set a global dns server:
```cpp
    esp_netif_dns_info_t dns;
    dns.ip.u_addr.ip4.addr = addr;
    dns.ip.type = IPADDR_TYPE_V4;
    ESP_ERROR_CHECK(esp_netif_set_dns_info(netif, ESP_NETIF_DNS_MAIN, &dns));
```
where, the `addr` would be the address of the google dns service.

david-cermak, 2021-05-28:
Closing, as this is a question (which has been answered), not an issue.

closed 2021-05-28 by david-cermak