espressif/esp-mqtt #192: MQTTS never connects. Example not working.

nikhilavrio, 2021-05-18:
Tried v4.2.1 and master branch esp-idf

Mosquitto Logs
```
621358243: OpenSSL Error[0]: error:0407008A:rsa routines:RSA_padding_check_PKCS1_type_1:invalid padding
1621358243: OpenSSL Error[1]: error:04067072:rsa routines:rsa_ossl_public_decrypt:padding check failed
1621358243: OpenSSL Error[2]: error:1417B07B:SSL routines:tls_process_cert_verify:bad signature
```

ESP32 Logs
```

I (95460) mbedtls: ssl_tls.c:3578 <= write record

I (95465) mbedtls: ssl_tls.c:3422 <= write handshake message

I (95472) mbedtls: ssl_cli.c:3656 <= write certificate verify

I (95479) mbedtls: ssl_cli.c:3785 client state: 10

I (95484) mbedtls: ssl_tls.c:2847 => flush output

I (95490) mbedtls: ssl_tls.c:2859 <= flush output

I (95495) mbedtls: ssl_tls.c:5987 => write change cipher spec

I (95502) mbedtls: ssl_tls.c:3286 => write handshake message

I (95508) mbedtls: ssl_tls.c:3445 => write record

I (95514) mbedtls: ssl_tls.c:2847 => flush output

I (95519) mbedtls: ssl_tls.c:2866 message length: 6, out_left: 6

I (95527) mbedtls: ssl_tls.c:2871 ssl->f_send() returned 6 (-0xfffffffa)

I (95534) mbedtls: ssl_tls.c:2899 <= flush output

I (95539) mbedtls: ssl_tls.c:3578 <= write record

I (95545) mbedtls: ssl_tls.c:3422 <= write handshake message

I (95551) mbedtls: ssl_tls.c:6001 <= write change cipher spec

I (95558) mbedtls: ssl_cli.c:3785 client state: 11

I (95564) mbedtls: ssl_tls.c:2847 => flush output

I (95569) mbedtls: ssl_tls.c:2859 <= flush output

I (95575) mbedtls: ssl_tls.c:6515 => write finished

I (95581) mbedtls: ssl_tls.c:6389 => calc  finished tls sha384

I (95589) mbedtls: ssl_tls.c:6419 <= calc  finished

I (95593) mbedtls: ssl_tls.c:3286 => write handshake message

I (95599) mbedtls: ssl_tls.c:3445 => write record

I (95605) mbedtls: ssl_tls.c:1446 => encrypt buf

I (95611) mbedtls: ssl_tls.c:1780 <= encrypt buf

I (95616) mbedtls: ssl_tls.c:2847 => flush output

I (95622) mbedtls: ssl_tls.c:2866 message length: 45, out_left: 45

I (95629) mbedtls: ssl_tls.c:2871 ssl->f_send() returned 45 (-0xffffffd3)

I (95636) mbedtls: ssl_tls.c:2899 <= flush output

I (95642) mbedtls: ssl_tls.c:3578 <= write record

I (95647) mbedtls: ssl_tls.c:3422 <= write handshake message

I (95654) mbedtls: ssl_tls.c:6624 <= write finished

I (95660) mbedtls: ssl_tls.c:2628 => fetch input

I (95665) mbedtls: ssl_tls.c:2789 in_left: 0, nb_want: 5

I (95671) mbedtls: ssl_tls.c:2813 in_left: 0, nb_want: 5

I (95677) mbedtls: ssl_tls.c:2814 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (95686) mbedtls: ssl_tls.c:2834 <= fetch input

I (95691) mbedtls: ssl_cli.c:3785 client state: 12

I (95697) mbedtls: ssl_tls.c:2847 => flush output

I (95702) mbedtls: ssl_tls.c:2859 <= flush output

I (95708) mbedtls: ssl_cli.c:3676 => parse new session ticket

I (95715) mbedtls: ssl_tls.c:4419 => read record

I (95720) mbedtls: ssl_tls.c:2628 => fetch input

I (95726) mbedtls: ssl_tls.c:2789 in_left: 5, nb_want: 5

I (95732) mbedtls: ssl_tls.c:2834 <= fetch input

I (95737) mbedtls: ssl_tls.c:2628 => fetch input

I (95742) mbedtls: ssl_tls.c:2789 in_left: 5, nb_want: 7

I (95749) mbedtls: ssl_tls.c:2813 in_left: 5, nb_want: 7

I (95755) mbedtls: ssl_tls.c:2814 ssl->f_recv(_timeout)() returned 2 (-0xfffffffe)

I (95763) mbedtls: ssl_tls.c:2834 <= fetch input

I (95769) mbedtls: ssl_tls.c:5278 got an alert message, type: [2:51]

W (95776) mbedtls: ssl_tls.c:5286 is a fatal alert message (msg 51)

W (95783) mbedtls: ssl_tls.c:4477 mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)

W (95793) mbedtls: ssl_cli.c:3680 mbedtls_ssl_read_record() returned -30592 (-0x7780)

I (95801) mbedtls: ssl_tls.c:8213 <= handshake

E (95807) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
I (95813) esp-tls-mbedtls: Certificate verified.
E (95819) esp-tls: Failed to open new connection
E (95824) TRANSPORT_BASE: Failed to open a new connection
E (95831) MQTT_CLIENT: Error transport connect
I (95835) MQTT_APP: MQTT_EVENT_ERROR
I (95839) MQTT_APP: Last error code reported from esp-tls: 0x8010
I (95846) MQTT_APP: Last tls stack error number: 0x7780
I (95852) MQTT_APP: MQTT_EVENT_DISCONNECTED
```

euripedesrocha, 2021-05-20:
Hi @nikhilavrio from broker's log and error reported by mbedtls,  seems that you have a problem in the certificate you are using to authenticate.

nikhilavrio, 2021-05-20:
Hi,

Yes, it is resolved. Certificates were the issue.

closed 2021-05-20 by nikhilavrio