espressif/esp-mqtt #195: W5500 & MQTT - Error transport connect

MadsHHLund, 2021-05-28:
I have a problem.

If I disconnect the cable for the Ethernet w5500 chip, then MQTT_Client does not reconnect.
I get the following error ...:
E (26603) TRANS_TCP: [sock = 54] select () timeout
E (26603) MQTT_CLIENT: Error transport connect.

If I use WIFI instead, and power down/up the WIFI connection, then a few seconds pass and the MQTT_Client connects beautifully again.

Should I look for the error in the W5500 driver, MQTT

By the way, I have adapted the W5500 driver, so that it supports the W6100 chip ...

david-cermak, 2021-05-28:
Hi @MadsHHLund 

There was a similar issue with the W5500 driver in IDF, on ethernet cable disconnection.
Linking the issue https://github.com/espressif/esp-idf/issues/6233 and the fix in https://github.com/espressif/esp-idf/commit/7af358ed93bf2f165bfc84341d1fb08f49f28573

Please check if your W6100 driver doesn't face the same problem.

MadsHHLund, 2021-05-28:
Hi David

I have added that fix Yesterday, but i didn’t help.

Med venlig hilsen

Mads Lund
Adm dir.

Danline Innovation A/S
Lumbyesvej 29 - 7000 Fredericia

Tlf.      +45 7620 6040
Dir.     +45 7620 6050
Mob. +45 2510 5888

***@***.******@***.***>  www.danline.dk<file:///var/tmp/com.apple.email.maild/EMContentRepresentation/com.apple.mobilemail/0C8101A1-CD35-407F-A3E7-D5C51A5F6317/www.danline.dk>

Linking the issue espressif/esp-idf#6233<https://github.com/espressif/esp-idf/issues/6233> and the fix in ***@***.***<https://github.com/espressif/esp-idf/commit/7af358ed93bf2f165bfc84341d1fb08f49f28573>

david-cermak, 2021-05-31:
@MadsHHLund  I don't think this is related to MQTT library.

Could you please check for ethernet and IP events? Will you get a new *Link-Up* event after reconnecting the cable? Will you also get a *Got-IP* event? Could you ping the device after reconnecting the ethernet cable? If you're seeing trouble with these, very likely would the issue be related to the ethernet driver (note that some other W5500 fixes have been merged to IDF master recently)

MadsHHLund, 2021-05-31:
Hi David

I think you are right…

I get the “Got-IP event” and Ip-address, but I can’t ping the device ??

Now I have a something to look for, thank you…

Med venlig hilsen

Mads Lund
Adm dir.

Danline Innovation A/S
Lumbyesvej 29 - 7000 Fredericia

Tlf.      +45 7620 6040
Dir.     +45 7620 6050
Mob. +45 2510 5888

[danline logo pc small small]

***@***.******@***.***>  www.danline.dk

MadsHHLund, 2021-05-31:
Hi David

Maybe you can help me further ...

As I wrote, I can not ping when the cable has been disconnected and reconnected.

If I completely comment out MQTT, then my software starts up, check via an HTTP call whether an OTA should be made.
Fetching time and date ...

and is then in an idle state, (here it would normally start MQTT up) ....

If I now pull the cable out, then

(27463) MAIN: Ethernet Link DISCONNECTED
I (33463) MAIN: Ethernet Link Up
I (33463) MAIN: Ethernet HW Addr 94: b9: 7e: 61: f6: 73
I (34193) esp_netif_handlers: eth ip: 172.16.2.123, mask: 255.255.255.0, gw: 172.16.2.1
I (34193) Ethernet.H: IP_EVENT_ETH_GOT_IP
I (34193) MAIN: MAIN LOOP IP Address 172.16.2.123
I (34193) MAIN: Name Server1: 172.16.2.41
I (34203) MAIN: Name Server2: 8.8.8.8
I (34203) MAIN: Name Server2: 8.8.8.8,

and I can ping the device, quite normally.

But when MQTT is with, it stops after Disconnect, Connect with

MQTT_EVENT_BEFORE_CONNECT:

and apparently fails at ::

if (esp_transport_connect (client-> transport,
                                      client-> config-> host,
                                      client-> config-> port,
                                      client-> config-> network_timeout_ms) <0) {
                ESP_LOGE (TAG, "Error transport connect");
                esp_mqtt_client_dispatch_tls_error (client);
                esp_mqtt_abort_connection (client);
                break;
            }

Unable to open a new socket.
Can't ping anymore.
Med venlig hilsen

Mads Lund
Adm dir.

Danline Innovation A/S
Lumbyesvej 29 - 7000 Fredericia

Tlf.      +45 7620 6040
Dir.     +45 7620 6050
Mob. +45 2510 5888

[danline logo pc small small]

***@***.******@***.***>  www.danline.dk

david-cermak, 2021-06-11:
Hi @MadsHHLund 

We have tested the MQTT library with W5500, paying close attention to disconnecting/reconnecting the ethernet cable. Everything works as expected, the ESP32 gets an IP event on reconnection and responds to pings normally, and reconnects to the MQTT broker.

```
I (4614) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
I (4614) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
E (8524) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (8524) MQTT_CLIENT: Poll read error: 119, aborting connection
I (8534) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (14004) esp_netif_handlers: example_connect: eth ip: 192.168.12.10, mask: 255.255.255.0, gw: 192.168.12.1
I (14004) example_connect: Got IPv4 event: Interface "example_connect: eth" address: 192.168.12.10
I (23534) MQTT_EXAMPLE: Other event id:7
I (23544) MQTT_EXAMPLE: MQTT_EVENT_CONNECTED
```

Please, note that it's been tested with the latest IDF `v4.4-dev` including these fixes to w5500 driver:
* https://github.com/espressif/esp-idf/commit/45846f06cffe98b7d5f09527df056c8e3d54a75e
* https://github.com/espressif/esp-idf/commit/0cd021adb1014f54683acfe69ce76f031c39c9b9
* https://github.com/espressif/esp-idf/commit/7af358ed93bf2f165bfc84341d1fb08f49f28573
* https://github.com/espressif/esp-idf/commit/dfe9580967a4b35871913ef3fdc47d8c6798f476

As for the issues you describe about not being able to start a new connection, ping or setup a new socket, please make sure you have enough heap space and/or deleting your connections and closing the sockets.

MadsHHLund, 2021-06-11:
Sorry - I should have answered earlier.

I have isolated my problem.
It occurred if, after receiving MQTT - disconnected, I tried to create a Reconnect.
When I removed all the "Disconnect / Connect" code, it beautifully made a reconnect, and now everything is running as it should.

My driver for the W6100 works fine - thanks for the help

closed 2021-06-11 by MadsHHLund