espressif/esp-mqtt #196: Arduino IDE compatibility

MrNoname3, 2021-05-30:
Hello!

Do you plan to make this library Arduino IDE and C++ compatible? It would be very useful, because I don't find any MQTT library which supports WSS with ESP32. PubSubClient library is very popular, but don't support WS/WSS. Maybe you could use it as source?

Thank you for your answer!

poerob, 2021-06-03:
I already run this library in an Arduino project. I'm using Development Version 2.0 of the Arduino core, i don't know if it's also included in the last stable version.
The minimal configuration example from [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html) works fine. 
`      const esp_mqtt_client_config_t mqtt_cfg = {
        .event_handle = NULL,
        .uri = "mqtt://Your_Broker_Uri.com",
        // .user_context = (void *)your_context
    };
    mqttClient = esp_mqtt_client_init(&mqtt_cfg); 
    esp_mqtt_client_register_event(mqttClient, MQTT_EVENT_CONNECTED, yourCallbackFunction, NULL);
    esp_mqtt_client_register_event(mqttClient, MQTT_EVENT_DISCONNECTED, yourCallbackFunction, NULL);
    esp_mqtt_client_start(mqttClient);`

david-cermak, 2021-06-11:
Hi @MrNoname3 

This library is supported in Arduino by default, as `esp-mqtt` is a public component of IDF (from v3.2) and the [stable arduino-esp32 release](https://github.com/espressif/arduino-esp32/releases/tag/1.0.6) is based on IDF `v3.3.5`.

The examples provided in IDF won't compile in C++, but it's very easy to adjust the code (removing designated initializes and casting `void*` of the event context ptr to an appropriate type). For the reference, you can check C++ test code in idf:

https://github.com/espressif/esp-idf/blob/master/tools/test_apps/protocols/mqtt/build_test/main/mqtt_app.cpp#L60

Attaching the simple sketch file, connecting over `wss` to a public mqtt broker, tested in Arduino IDE (v. 1.8.15 / v. 2.0 beta):
[ARDUINO_MQTT_WS.ino.txt](https://github.com/espressif/esp-mqtt/files/6636463/ARDUINO_MQTT_WS.ino.txt)

Closing, as this is not an issue.

closed 2021-06-11 by david-cermak