espressif/esp-mqtt #200: Access QoS on receiving events

bertmelis, 2021-09-02:
Is it possible to get the QoS when receiving messages or subacks?

I can't find any docs on this. Happy to create a PR if not yet implemented. (note: I did not try to implement yet)

david-cermak, 2021-09-07:
@bertmelis Only the retain flag is available in the data event:

https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L958

[qos and dup flags](https://github.com/espressif/esp-mqtt/blob/master/lib/include/mqtt_msg.h#L113-L124) could be added in a similar way. PRs welcome :+1:

bertmelis, 2021-09-13:
While implementing I overlooked something I also wanted: the payload from the SUBACK message.
This payload contains the maximum granted qos level of the subscribe, or 0xF for failure.

OK to put in the same PR?

david-cermak, 2021-09-14:
> OK to put in the same PR?

Absolutely, but please make it a separate commit.

closed 2021-10-04 by david-cermak