espressif/esp-mqtt #201: esp_mqtt_connect: mqtt_message_receive() returned -1 , MQTT connect failed

torntrousers, 2021-09-04:
I'm trying to use esp-mqtt with mqtt over websockets connecting to AWS IoT with a custom authorizer, but it fails to connect. The request gets in to AWS ok and I can see the custom authoriser has run and and authorised the connection. On the ESP32 debug logs I get:

```
E (302789) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
E (302790) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (302794) MQTT_CLIENT: MQTT connect failed
```

Any tips on how I could debug this ?

TIA!

euripedesrocha, 2021-09-07:
Hi @torntrousers, could you please run your code with:

```
 esp_log_level_set("MQTT_CLIENT", ESP_LOG_DEBUG);
```
and share the logs with us.

torntrousers, 2021-09-08:
Sure, sorry its going to take a couple of days to get back to that code...

torntrousers, 2021-09-21:
I'm sorry but I can't find where I have put the code for testing this, so I will close for now.

closed 2021-09-21 by torntrousers