espressif/esp-mqtt #202: Already connected when got MQTT_EVENT_BEFORE_CONNECT event

AxelLin, 2021-09-12:
The mqtt library post MQTT_EVENT_BEFORE_CONNECT before connecting.
But it does not wait for the task that receiving MQTT_EVENT_BEFORE_CONNECT done.
Which means when you got MQTT_EVENT_BEFORE_CONNECT it's possible too late for re-configuration.
https://github.com/espressif/esp-mqtt/b ... 1367-L1388

So if the user wants to re-config settings before connecting the broker and put the code in MQTT_EVENT_BEFORE_CONNECT event handler, it's too late because the code may be executed **after** connected.

euripedesrocha, 2021-09-13:
Hi @AxelLin,  thanks for reporting this.

If you go deep and follow the event dispatch process you will see that we don't have a dedicated task for the event loop, therefore all dispatched events will run immediately on MQTT task context and the scenario you described will not occur.

AxelLin, 2021-09-13:
https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L1369 calls esp_mqtt_dispatch_event_with_msgid(client);
https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L1379 calls esp_transport_connect()

Application code calls esp_mqtt_client_register_event(mqtt_client, MQTT_EVENT_ANY, mqtt_event_handler, NULL);
There is no guarantee that the code in mqtt_event_handler when receive MQTT_EVENT_BEFORE_CONNECT will be executed **before** https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L1379.

MartyMcFlyInTheSky, 2021-09-14:
I think AxelLin is right about his assumption. It is assumed that the event loop will run faster than esp_transport_connect() but there's no guarantee that it will.

euripedesrocha, 2021-09-14:
@AxelLin [esp_mqtt_dispatch_event_with_msgid](https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L910) is blocking, not async as you may expect by reading the code. 

If the user register the event after the client starts, naturally the before connect will not be called. This would be an error on the API usage by the user. 

The whole process is [here](https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L910-L931). You can see [esp_event_loop_run call here](esp_event_loop_run). 

In summary using the event loop has the same behavior of the usage of the event handler as a callback in the config struct and is provided as interface for better consistency with other idf components. 

@MartyMcFlyInTheSky as I wrote above the event loop runs in the dispatch process, so it's guaranteed to run before the call to `esp_transport_connect`.

AxelLin, 2021-09-14:
@euripedesrocha 

I think you still don't get the point.

Take a look of the example in esp-idf:

https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/tcp/main/app_main.c#L142-L143
https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/tcp/main/app_main.c#L53-L108

If the mqtt library post an MQTT_EVENT_BEFORE_CONNECT event, when the code in mqtt_event_handler() received the event
it's actually already connected.

AxelLin, 2021-09-14:
@euripedesrocha 

Oops, my bad. I test again and it seems fine now.
(Could be something wrong when I tested with old version before MQTT_SUPPORTED_FEATURE_EVENT_LOOP was added).

closed 2021-09-14 by AxelLin

euripedesrocha, 2021-09-14:
@AxelLin, could you elaborate on how is this possible?

`MQTT_EVENT_BEFORE_CONNECT` is only posted by the library in `MQTT_STATE_INIT` [here](https://github.com/espressif/esp-mqtt/blob/026ea9533896eafc88a0545c51a799f6cdf5f508/mqtt_client.c#L1359), and the handler is guaranteed to be called and run before `esp_transport_connect` is called as I explained in my previous answer.

AxelLin, 2021-09-14:
> @AxelLin, could you elaborate on how is this possible?
> 

Last time I tested the MQTT_EVENT_BEFORE_CONNECT event (quite long time ago) and found I usually got MQTT_EVENT_BEFORE_CONNECT **after** connected.
But I don't recall which esp-idf version and esp-mqtt version was tested.
If I recall correctly at that time I don't see this line in mqtt_clent.c: return esp_event_loop_run(client->config->event_loop_handle, 0); (But my memory could be wrong anyway).

AxelLin, 2021-09-15:
@euripedesrocha 

This probably off-topic, but I'm wondering what is the benefit of using MQTT_SUPPORTED_FEATURE_EVENT_LOOP instead of setting client->config->event_handle?

i.e.
https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L921-L922 v.s
https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L925-L926

euripedesrocha, 2021-09-15:
@AxelLin the benefit is to have a consistent API with other IDF components.

AxelLin, 2021-09-17:
> @AxelLin the benefit is to have a consistent API with other IDF components.

I don't get it.
This is about internal implementation of esp_mqtt_dispatch_event() which is nothing to do with API things.
What make me confused is why post an EVENT to the same task? Isn't it simpler by just calling client->config->event_handle?