espressif/esp-mqtt #205: MQTT client not responding

goofy2k, 2021-09-23:
Hi, I am trying to use the esp_mqtt example for tcp.  I use ESP-IDF from the command line under Windows. Have modified the example for compilation as C++. After solving initialization of the MQTT client (see closed issue #204) I get no logging response from the client at all. I try to connect with my local broker. I am sure that this broker is up and running OK as I use it also in other applications, with the same connection details.  How can I go on to solve this?

This is my initialization code, with commented left-overs of my trials:

'''
static void mqtt_app_start(void){
    //esp_mqtt_client_config_t mqtt_cfg = {0};
    esp_mqtt_client_config_t mqtt_cfg = {};
    // mqtt_cfg.uri = CONFIG_BROKER_URL;
    mqtt_cfg.uri = "mqtt://192.168.2.200:1883";
    //strcpy((char*)mqtt_cfg.username, "Fred");
    mqtt_cfg.username = "Fred";
    //strcpy((char*)mqtt_cfg.password, "xxxxxxxxxx");
    mqtt_cfg.password = "xxxxxxxxxx";
    //strcpy((char*)mqtt_cfg.client_id, "TTGO_TAudio_1");
    mqtt_cfg.client_id = "TTGO_TAudio_1";

    ......

    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
    mqtt_cfg.event_handle = mqtt_event_handler_cb;
    //esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
    ESP_LOGI(TAG, "START MQTT CLIENT");  //FCKX
    esp_mqtt_client_start(client);
    ESP_LOGI(TAG, "MQTT CLIENT STARTED");  //FCKX
   ....
'''

Here is the output of my app:
'''
I (563) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (574) MQTT_EXAMPLE: [APP] Startup..
I (574) MQTT_EXAMPLE: [APP] Free memory: 277916 bytes
I (584) MQTT_EXAMPLE: [APP] IDF version: v4.2-dirty
I (644) wifi:wifi driver task: 3ffc04f0, prio:23, stack:6656, core=0
I (644) system_api: Base MAC address is not set
I (644) system_api: read default base MAC address from EFUSE
I (664) wifi:wifi firmware version: 1865b55
I (664) wifi:wifi certification version: v7.0
I (664) wifi:config NVS flash: enabled
I (664) wifi:config nano formating: disabled
I (674) wifi:Init data frame dynamic rx buffer num: 32
I (674) wifi:Init management frame dynamic rx buffer num: 32
I (684) wifi:Init management short buffer num: 32
I (684) wifi:Init dynamic tx buffer num: 32
I (694) wifi:Init static rx buffer size: 1600
I (694) wifi:Init static rx buffer num: 10
I (704) wifi:Init dynamic rx buffer num: 32
I (704) wifi_init: rx ba win: 6
I (704) wifi_init: tcpip mbox: 32
I (714) wifi_init: udp mbox: 6
I (714) wifi_init: tcp mbox: 6
I (714) wifi_init: tcp tx win: 5744
I (724) wifi_init: tcp rx win: 5744
I (724) wifi_init: tcp mss: 1440
I (734) wifi_init: WiFi IRAM OP enabled
I (734) wifi_init: WiFi RX IRAM OP enabled
I (744) example_connect: Connecting to Verhoeckx_glas...
I (834) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (834) wifi:mode : sta (84:cc:a8:0d:3b:b8)
I (844) example_connect: Waiting for IP(s)
I (964) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:1
I (964) wifi:state: init -> auth (b0)
I (984) wifi:state: auth -> assoc (0)
I (1034) wifi:state: assoc -> run (10)
I (1134) wifi:connected with yyyyyyyyyy, aid = 4, channel 8, BW20, bssid = 96:6a:b0:c3:3b:27
I (1134) wifi:security: WPA2-PSK, phy: bgn, rssi: -49
I (1144) wifi:pm start, type: 1
I (1164) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2624) esp_netif_handlers: example_connect: sta ip: 192.168.2.16, mask: 255.255.255.0, gw: 192.168.2.254
I (2624) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.2.16
I (2634) example_connect: Connected to example_connect: sta
I (2634) example_connect: - IPv4 address: 192.168.2.16
I (2644) MQTT_EXAMPLE: START MQTT CLIENT
I (2644) MQTT_EXAMPLE: MQTT CLIENT STARTED
'''

david-cermak, 2021-09-23:
Hi @goofy2k 

Could you please enable debug logs in the menuconfig?
```
(Top) → Component config → Log output → Default log verbosity
```
Also I'd suggest registering an event handler using

https://github.com/espressif/esp-idf/blob/master/tools/test_apps/protocols/mqtt/build_test/main/mqtt_app.cpp#L63

And logging at least the event id in the handler.
After starting the client, you should get at least `MQTT_EVENT_BEFORE_CONNECT` event.

This initialization should be enough to connect to a local broker (plain tcp, no security):
```cpp
    esp_mqtt_client_config_t mqtt_cfg = {0};
    mqtt_cfg.uri = "mqtt://192.168.xx.xx:1883";
    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(client, MQTT_EVENT_ANY, mqtt_event_handler, client);
    esp_mqtt_client_start(client);
```

goofy2k, 2021-09-23:
Going through esp_mqtt.c   by placing ESP_LOGI ' s
Trying to detect where it is stuck....

goofy2k, 2021-09-23:
Thanks @david_cermak,  I'll go into this! I have less time tomorrow, so be
patient ;-)

Fred

goofy2k, 2021-09-24:
I now have DEBUG level logging ON and get some more information.
At the moment I don't register the client as you suggest. It gave me errors at compilation time (I use main.cpp !) . I found a suggestion by you in the closed issue that I mentioned to use  mqtt_cfg.event_handle = mqtt_event_handler_cb; and remove this: static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data) .
I'll send you my code  and the output in a while.

goofy2k, 2021-09-24:
Also note that I do NOT maintain esp-idf under git.  So I may not have the latest updates to mqtt_client.c    .  I did not do this because I'm new to git/Github.

goofy2k, 2021-09-24:
My output:

It takes quite a while (30 seconds?) before SENT ping succesful appears. 

I (1540) wifi:security: WPA2-PSK, phy: bgn, rssi: -47
I (1540) wifi:pm start, type: 1

I (1640) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2960) esp_netif_handlers: example_connect: sta ip: 192.168.2.16, mask: 255.255.255.0, gw: 192.168.2.254
I (2960) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.2.16
I (2970) example_connect: Connected to example_connect: sta
I (2970) example_connect: - IPv4 address: 192.168.2.16
D (2980) MQTT_CLIENT: MQTT client_id=TTGO_TAudio_1
I (2990) MQTT_EXAMPLE: START MQTT CLIENT
D (2990) MQTT_CLIENT: Core selection disabled
I (2990) MQTT_EXAMPLE: MQTT CLIENT STARTED
D (3010) MQTT_CLIENT: Transport connected to mqtt://192.168.2.200:1883
D (3010) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (3020) MQTT_CLIENT: mqtt_message_receive: first byte: 0x20
D (3020) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (3030) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (3040) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (3040) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (3050) MQTT_CLIENT: Connected
D (63050) MQTT_CLIENT: Sent PING successful
D (63050) MQTT_CLIENT: PING sent
D (63050) MQTT_CLIENT: mqtt_message_receive: first byte: 0xd0
D (63060) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x0
D (63060) MQTT_CLIENT: mqtt_message_receive: total message length: 2 (already read: 2)
D (63070) MQTT_CLIENT: mqtt_message_receive: transport_read():2 2
D (63080) MQTT_CLIENT: msg_type=13, msg_id=0
D (63080) MQTT_CLIENT: MQTT_MSG_TYPE_PINGRESP

goofy2k, 2021-09-24:
The code:

    esp_mqtt_client_config_t mqtt_cfg = {0};
    // mqtt_cfg.uri = CONFIG_BROKER_URL;
    //mqtt_cfg.uri = "mqtt://external broker ip";
    mqtt_cfg.uri = "mqtt://192.168.2.200:1883";              //use local ip for now
    //strcpy((char*)mqtt_cfg.uri, "mqtt://external broker ip");   //contains the external ip to my broker..... ;-)
    //strcpy((char*)mqtt_cfg.username, "Fred");
    mqtt_cfg.username = "Fred";
    //strcpy((char*)mqtt_cfg.password, "xxxxxxxxxx);
    mqtt_cfg.password = "Nwwnlil_12";
    //strcpy((char*)mqtt_cfg.client_id, "TTGO_TAudio_1");
    mqtt_cfg.client_id = "TTGO_TAudio_1";

    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
    mqtt_cfg.event_handle = mqtt_event_handler_cb;
    //esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);  //this gives error at compilation (as C++)
    ESP_LOGI(TAG, "START MQTT CLIENT");  //FCKX
    esp_mqtt_client_start(client);
    ESP_LOGI(TAG, "MQTT CLIENT STARTED");  //FCKX

goofy2k, 2021-09-24:
The commented lines are from trials, when I had to deal with the compilation errors....

goofy2k, 2021-09-24:
I have another MQTT client running (under Nodered). I could detect any traffic there. I subscribed to this topic:   topic/qos0  . As I understand from the example code that it will publish to that. No results....

goofy2k, 2021-09-24:
Hm,  I have this ESP_LOGI(TAG, "HANDLER_CB CALLED");  //FCKX  in the top of the handler_cb.  Don't see that message in my output.....

goofy2k, 2021-09-24:
I've got something running!  MQTT messages arrive.  I'll let you know later what I did....

goofy2k, 2021-09-24:
I now use this:    

esp_mqtt_client_register_event(client, (esp_mqtt_event_id_t)ESP_EVENT_ANY_ID, (esp_event_handler_t)mqtt_event_handler, client);  //trial

had to add type casting two times ...

goofy2k, 2021-09-28:
This was the solution for my issue, so I propose to close the issue

closed 2021-09-28 by goofy2k