espressif/esp-mqtt #206: reconnect not callable from event

bertmelis, 2021-09-27:
I have autoreconnect disabled.

Am I correct that you cannot call `esp_mqtt_client_reconnect` from within the event handler on `MQTT_EVENT_DISCONNECTED` events? I get a `ESP_FAIL` error because the clients isn't ready yet.

Is this intentional?

david-cermak, 2021-10-03:
@bertmelis Thanks for the report, this is definitely a faulty behavior.

> Is this intentional?

It was *intentional* in the past, as `disable_auto_reconnect` was introduced after manual reconnection, so it was supposed to work properly, only if autoreconnect been disabled by setting the `reconnect_timeout_ms` a very-very big number. But now we have to update the reconnection logic.

The WIP fix is below:
[Client-Fix-reconnect-disconnect-if-auto_reconnect-of.patch.txt](https://github.com/espressif/esp-mqtt/files/7272904/Client-Fix-reconnect-disconnect-if-auto_reconnect-of.patch.txt)

This works for me in this connect-disconnect scenario:
```cpp
    switch (event_id) {
    case MQTT_EVENT_CONNECTED:
        esp_mqtt_client_disconnect(client);
        break;
    case MQTT_EVENT_DISCONNECTED:
        esp_mqtt_client_reconnect(client);
        break;
    default:
        break;
    }
```

We're planning on adding more unit tests to cover these basic state transitions to avoid happening this in future.

bertmelis, 2021-10-03:
Thanks. I'll try with the patch.

Just a question: are unit tests you're talking about public? In this repo CI is only to check whether the example builds, right?

david-cermak, 2021-10-03:
> are unit tests you're talking about public?

Most of the mqtt tests are located in IDF, but, yes, they are public. As for the unit tests we have a very simple suite of
* target tests: https://github.com/espressif/esp-idf/blob/master/components/mqtt/test/test_mqtt.c
* host tests: https://github.com/espressif/esp-idf/tree/master/components/mqtt/host_test

> In this repo CI is only to check whether the example builds

This GitHub repo runs a very basic `travis` build only. In addition to that we run a simple CI pipeline on the internal mirror, basically building against multiple IDF versions and running a smoke test checking for common connection/publishing scenarios:

https://github.com/espressif/esp-mqtt/blob/master/.gitlab-ci.yml#L122

(again running an [IDF test app](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/protocols/mqtt/publish_connect_test#esp-mqtt-advanced-publish-and-connect-test-project) )