espressif/esp-mqtt #208: esp_mqtt_client_disconnect should send DISCONNECT Packet before close session

AxelLin, 2021-09-30:
[1] 3.14 DISCONNECT – Disconnect notification
The DISCONNECT Packet is the final Control Packet sent from the Client to the Server.
It indicates that the Client is disconnecting cleanly.

Calling esp_mqtt_client_disconnect() does not send DISCONNECT Packet, it just close the tcp session.
(Only esp_mqtt_client_stop send DISCONNECT Packet, this seems wrong.)

[1] http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718090

bertmelis, 2021-09-30:
Would this mean that when disconnected the last will is always sent even when we gracefully disconnect.

david-cermak, 2021-10-03:
@AxelLin Thanks for the report, this needs fixing.

This patch should fix the problem:
[Send-disconnect-message-if-client-asked-to-disconnec.patch.txt](https://github.com/espressif/esp-mqtt/files/7272912/Send-disconnect-message-if-client-asked-to-disconnec.patch.txt)

@bertmelis Yes, whenever a user app called `esp_mqtt_client_disconnect()` the connection would be abruptly terminated and lwt would be published (with current esp-mqtt master)

AxelLin, 2021-11-29:
> fixed by [86e40f8](https://github.com/espressif/esp-mqtt/commit/86e40f86152a522b179b5dcc1dbbd0b288ee8321)

However, the esp-idf does not update esp-mqtt submodule.
So this is actually not (yet) fixed in any esp-idf branch now.