espressif/esp-mqtt #210: Question: event: running post MQTT_EVENTS: 6 with handler.... and context .... on loop ...

goofy2k, 2021-11-03:
Hi, just a question.  I see messages as indicated in the issue title in my logs.  What does it mean?

AndriiFilippov, 2021-11-04:
Hi, @goofy2k ! 

This is a debugging log from the esp_event. 
https://github.com/espressif/esp-idf/blob/master/components/esp_event/esp_event.c#L127

goofy2k, 2021-11-04:
OK, thanks Andrii,  so there is nothing wrong :-). I just have to lower the
logging setting to less than DEBUG.

Alvin1Zhang, 2021-11-05:
Thanks for reporting, you could also create a post on our forum at https://esp32.com, thanks.

david-cermak, 2021-11-05:
@goofy2k I agree with @Alvin1Zhang , please prefer the forum for basic/general questions about IDF/ESP32(-X).

anyway, I think you're satisfied with @AndriiFilippov 's answers, so closing this issue.

closed 2021-11-05 by david-cermak