espressif/esp-mqtt #213: Sometimes loop stop forever after: MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0

garudaonekh, 2021-11-29:
Hi,
My code sometimes stop running after MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0, i.e, it can't return from this function: esp_mqtt_client_publish. This happen after several mqtt_client timeout.

I think my issue is related with this: https://github.com/espressif/esp-mqtt/issues/126

My code is mainly based on this: https://github.com/levkovigor/MQTT_VPN_ESP32/blob/main/mqtt_vpn_webserver/mqttif.c

Any suggestion, thanks?

david-cermak, 2021-12-15:
@garudaonekh  Could you please provide a simple project that would exhibit his problem?

david-cermak, 2022-08-18:
Closing for short of feedback. Please reopen if needed with more details.

closed 2022-08-18 by david-cermak