espressif/esp-mqtt #214: esp_mqtt_client_publish with data longer than buffer causes disconnect and issues with Mosquitto

DCSBL, 2021-12-20:
# Summary
When using `esp_mqtt_client_publish` with a message longer than the buffer size (`MQTT_BUFFER_SIZE_BYTE`), the client disconnects and Mosquitto receives garbage data.

# Environment
- Tested with ESP-IDF: `v4.1.2`, `v4.3` and `[master]` @ (https://github.com/espressif/esp-idf/commit/279c8aeb8a312a178c73cf4b50a30798332ea79b), (Issue not present on `v4.1`)
- Tested with selected esp-mqtt submobule and with https://github.com/espressif/esp-mqtt/commit/383e7e5f0f3e1af4530e01a2d784ee83110594d8.
- ESP32-WROOM-32D
- Mosquitto `2.0.14` with this config:

    ```
    allow_anonymous true

    listener 1883
    protocol mqtt

    listener 80
    protocol websockets
    ```

# How to reproduce
1. Used this example https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ws @ https://github.com/espressif/esp-idf/commit/279c8aeb8a312a178c73cf4b50a30798332ea79b
2. Configured example project with Wifi SSID and pass and MQTT server address.
3. Added this at line 63 (after sending `topic/qos0`/`"data"` (string with +1200 characters):
    ```diff
   case MQTT_EVENT_SUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
        msg_id = esp_mqtt_client_publish(client, "/topic/qos0", "data", 0, 0, 0);
        ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
        
    +   msg_id = esp_mqtt_client_publish(client, "/topic/long", "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Suspendisse nibh nibh, tempus ut vulputate in, auctor eu quam. Suspendisse vel auctor nisi. Nam tellus erat, porta auctor vulputate vel, venenatis id arcu. Phasellus a viverra risus. Maecenas imperdiet nibh augue, ut malesuada sapien scelerisque eget. Donec iaculis dui ut ultrices interdum. Maecenas convallis mattis lobortis. Quisque ac congue ipsum, eu mollis est. Pellentesque nec ante leo. Nam a ipsum tristique, sagittis nunc non, mollis sapien. Cras quam urna, consectetur ut tellus vulputate, venenatis suscipit mi. Proin pretium arcu convallis, dictum dolor vitae, ultricies nulla. Integer id eleifend mauris. Integer iaculis, augue ut mattis tempor, est felis sodales justo, ac interdum lacus orci non eros. Nulla accumsan ut arcu sed ornare. Ut aliquet erat vel eleifend consequat. Ut sit amet tincidunt eros. Vivamus nec libero ut velit pretium molestie. Orci varius natoque penatibus et magnis dis parturient montes, nascetur ridiculus mus. Nullam rhoncus nisi ut interdum blandit. Fusce tincidunt dictum tellus, sit amet suscipit ex fermentum tempus. Integer feugiat iaculis volutpat. Pellentesque habitant morbi tristique posuere.", 0, 0, 0);
    +   ESP_LOGI(TAG, "sent long publish successful, msg_id=%d", msg_id);
        break;
    ```
4. Uploaded to ESP, got the following result:
    <details>
    <summary>Click to see full verbose log</summary>
    
    ```
    I (5571) esp_netif_handlers: example_connect: sta ip: 192.168.5.126, mask: 255.255.255.0, gw: 192.168.5.1
    I (5571) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.5.126
    I (5581) example_connect: Connected to example_connect: sta
    I (5581) example_connect: - IPv4 address: 192.168.5.126
    D (5591) MQTT_CLIENT: MQTT client_id=ESP32_2547D8
    D (5591) MQTT_CLIENT: Core selection disabled
    I (5601) MQTTWS_EXAMPLE: Other event id:7
    W (5641) wifi:<ba-add>idx:0 (ifx:0, 78:8a:20:8d:ec:be), tid:0, ssn:0, winSize:64
    D (5651) TRANSPORT_WS: sub_protocol: mqtt
    D (5651) TRANSPORT_WS: Write upgrade request
    GET / HTTP/1.1
    Connection: Upgrade
    Host: 192.168.5.119:80
    User-Agent: ESP32 Websocket Client
    Upgrade: websocket
    Sec-WebSocket-Version: 13
    Sec-WebSocket-Key: TDRus8asdaEKkZPR4P+T9Qg==
    Sec-WebSocket-Protocol: mqtt
    
    
    D (5801) TRANSPORT_WS: Read header chunk 159, current header size: 159
    D (5801) TRANSPORT_WS: server key=1kZasdd2xa++p1OP/LEiPTNOxmrI=, send_key=TDRus8asdaEKkZPR4P+T9Qg==,     expected_server_key=1kZasdd2xa++p1OP/LEiPTNOxmrI=
    D (5811) MQTT_CLIENT: Transport connected to ws://192.168.5.119:80
    D (5811) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
    D (5861) TRANSPORT_WS: Opcode: 2, mask: 0, len: 4
    
    D (5861) TRANSPORT_WS: Actual data to receive (4) are longer than ws buffer (1)
    D (5861) MQTT_CLIENT: mqtt_message_receive: first byte: 0x20
    D (5871) TRANSPORT_WS: Actual data to receive (3) are longer than ws buffer (1)
    D (5871) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
    D (5881) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
    D (5891) MQTT_CLIENT: mqtt_message_receive: read_len=2
[... 121 lines cut ...]
    I (6044) MQTTWS_EXAMPLE: sent long publish successful, msg_id=0
    I (6054) MQTTWS_EXAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=20415
    I (6054) MQTTWS_EXAMPLE: sent publish successful, msg_id=0
    I (6064) MQTTWS_EXAMPLE: sent long publish successful, msg_id=0
    I (6074) MQTTWS_EXAMPLE: MQTT_EVENT_UNSUBSCRIBED, msg_id=57707
    I (6074) MQTTWS_EXAMPLE: MQTT_EVENT_DATA
    TOPIC=/topic/qos0
    DATA=data
    E (6084) TRANSPORT_WS: Error read data
    E (6084) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
    I (6094) MQTTWS_EXAMPLE: MQTT_EVENT_ERROR
    E (6094) MQTTWS_EXAMPLE: Last error reported from esp-tls: 0x8008
    I (6104) MQTTWS_EXAMPLE: Last errno string (Success)
    E (6104) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
    I (6114) MQTTWS_EXAMPLE: MQTT_EVENT_DISCONNECTED
    ```

I've also subscribed to the topics to see the what is send (`mosquitto_sub -v -h 192.168.5.119 -t '#'`). Here you see that there is some weird characters being send (I think a missing null terminator). 
```
/topic/qos1 data_3
/topic/qos0 data
/topic/long Lorem ipsum dolor sit amet, consectetur adipiscing elit. Suspendisse nibh nibh, tempus ut vulputate in, auctor eu quam. Suspendisse vel auctor nisi. Nam tellus erat, porta auctor vulputate vel, venenatis id arcu. Phasellus a viverra risus. Maecenas imperdiet nibh augue, ut malesuada sapien scelerisque eget. Donec iaculis dui ut ultrices interdum. Maecenas convallis mattis lobortis. Quisque ac congue ipsum, eu mollis est. Pellentesque nec ante leo. Nam a ipsum tristique, sagittis nunc non, mollis sapien. Cras quam urna, consectetur ut tellus vulputate, venenatis suscipit mi. Proin pretium arcu convallis, dictum dolor vitae, ultricies nulla. Integer id eleifend mauris. Integer iaculis, augue ut mattis tempor, est felis sodales justo, ac interdum lacus orci non eros. Nulla accumsan ut arcu sed ornare. Ut aliquet erat vel eleifend consequat. Ut sit amet tincidunt eros. Vivamus nec libero ut velit pretium molestie. Orci varius natoque penatibus et magnis dis parturient montes, nascetur ridiculus 0
                                                   /topic/qos0data0�
                                                                        /topic/longLorem ipsum dolor sit amet, consectetur adipiscing elit. Suspendisse nibh nibh, tempus ut vulputate in, auctor eu quam. Suspendisse vel auctor nisi. Nam tellus e
```

ESP-YJM, 2021-12-21:
@DCSBL Please use this patch and test
[0001-mqtt_client-Fix-mqtt-send-long-data-error.zip](https://github.com/espressif/esp-mqtt/files/7750497/0001-mqtt_client-Fix-mqtt-send-long-data-error.zip)
.

DCSBL, 2021-12-21:
Thanks. Tested with same example and https://github.com/espressif/esp-idf/commit/279c8aeb8a312a178c73cf4b50a30798332ea79b. Confirmed that this patch fixes my issue.

DCSBL, 2021-12-21:
I am not really sure, but it seems that [this test](https://github.com/espressif/esp-idf/blob/master/tools/test_apps/protocols/mqtt/publish_connect_test/app_test.py) does not tests payloads larger than the buffer size. The largest payload is tested with `start_publish_case(transport, qos, 50, 1, q)`: `50 * DEFAULT_MSG_SIZE (16) = 800`, smaller than `MQTT_BUFFER_SIZE_BYTE` (1024 by default).

Is it possible to test for this to prevent something like this issue?

ESP-YJM, 2021-12-21:
@DCSBL Thanks for pointing out it. I think it is a good idea, we will implement it.

closed 2021-12-22 by david-cermak