espressif/esp-mqtt #215: Unable to use global ca store

brahma-dev, 2021-12-22:
I am trying to use the global ca store as follows.

```c
const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://example.com",
		.use_global_ca_store = true,
		.crt_bundle_attach = esp_crt_bundle_attach
    };
```
But it appears that `use_global_ca_store` and `crt_bundle_attach` can't be used together, which results in an empty global cert. `esp-tls-mbedtls: global_cacert is NULL`

https://github.com/espressif/esp-mqtt/blob/026253d802473ac9afd4cc747e2632e96f71acbb/mqtt_client.c#L205-L207

euripedesrocha, 2022-01-10:
Hi @brahma-dev,
you are correct, these options are mutually exclusive. 

Could you elaborate on why you want to use both?

euripedesrocha, 2022-01-10:
I suggest that you first take a look on [esp-tls documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html?highlight=global_ca_store) about the usage of the `global_ca_store`

brahma-dev, 2022-01-11:
Apologies and thanks, I had misunderstood `use_global_ca_store`.

closed 2022-01-11 by brahma-dev