espressif/esp-mqtt #218: Simultaneous MQTT and HTTP server

SERIDJ, 2022-01-07:
hi, 

is it possible to run using 2 task an MQTT client (port 8883) to publish data and have web page at same time .
for now each task work well separately. But i got error when i try to run both task at same code.

like : 

 E (385828) TRANS_TCP: tcp_poll_read select error 113, errno = Software caused connection abort, fd = 57
E (385848) MQTT_CLIENT: Poll read error: 119, aborting connection.
W (385848) httpd: httpd_accept_conn: error in accept (128).
E (395888) TRANS_TCP: [sock=54] connect() error: Host is unreachable
E (395888) MQTT_CLIENT: Error transport connect

david-cermak, 2022-01-10:
Hi @SERIDJ 

It seems like you might be running out of available sockets in your system. Please check the settings of [`CONFIG_LWIP_MAX_SOCKETS`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-max-sockets)

No problem in general to run multiple instances of mqtt client and http server at the same time. Things to consider: Enough socket descriptors, enough memory, each instance of the mqtt client should have a different client id.

SERIDJ, 2022-01-10:
@david-cermak  thanks for ur replay 

 yes i fixe it , its probleme of stack_size  ... but how i can calcule the exact size for each task, i know its by word so esp32 is 32 bit so one word is 4 bytes ....

david-cermak, 2022-01-10:
> yes i fixe it , its probleme of stack_size

Ok, thanks for the update. Closing this.

> how i can calcule the exact size 

While It is theoretically possible to calculate the needed stack depth, I'd suggest trying this API to get the maximum runtime stack usage [uxTaskGetStackHighWaterMark](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv427uxTaskGetStackHighWaterMark12TaskHandle_t) while running your app for some time.

closed 2022-01-10 by david-cermak