espressif/esp-mqtt #221: [BUG] 使用AT指令AT+MQTTCONN返回ERROR的错误

iAoe444, 2022-04-14:
## 遇到的问题

使用AT指令`AT+MQTTCONN`返回错误ERROR, 错误码是`0x600B`

## 出现问题的版本

编译的esp-at版本：`16236bc`

## 错误LOG如下

```
all parameters are ready[0;33mW (561901) MQTT_CLIENT: Transport config set, but overridden by scheme from URI: transport = 1, uri scheme = mqtt[0m
[0;33mW (561902) MQTT_CLIENT: Client asked to stop, but was not started[0m
[0;31mE (561909) MQTT_CLIENT: Client was not initialized[0m
[at-mqtt]error code:0x600B[at-mqtt]error code:0x6030
```

## 个人排查过程

1. 通过添加打印日志的方式，发现问题出现在mqtt_client.c的第340行进行的返回
 
```
  if (cfg->uri && user_cfg->transport) {
      ESP_LOGW(TAG, "Transport config set, but overridden by scheme from URI: transport = %d, uri scheme = %s", user_cfg->transport, cfg->scheme);
      ret = ESP_ERR_INVALID_ARG;
  }

  return ret;
```

此时 cfg->uri && user_cfg->transport 判断为真，导致mqtt创建不成功返回错误，通过查看代码，个人觉得`cfg->uri && user_cfg->transport` 在正确的情况也会返回为真，最后会返回ESP_ERR_INVALID_ARG，导致AT+MQTTCONN返回错误

2. 之前烧录过`v2.2.0.0_esp32`的代码，发现没有这个问题，于是翻看了在这个时间段之间做的修改，发现在`1d10608`时合并了代码进来，之前代码固定返回是ESP_OK，所以没有在`v2.2.0.0_esp32`之前没有这个问题，在最新的代码才有这个问题

3. 所以个人觉得这个算是一个BUG的存在

ustccw, 2022-04-15:
@iAoe444 esp-at project has already fixed this in a workaround way several weeks ago. could you please try with the latest esp-at firmware.

ustccw, 2022-04-15:
it should get fixed on https://github.com/espressif/esp-at/commit/ed9e3e7f56e63dba55a2db80e583ebb99df09bae

closed 2022-06-03 by david-cermak