espressif/esp-mqtt #225: Keepalive failed after updating system time via SNTP

ser90368141, 2022-05-23:
When updating the system time via SNTP, the keepalive timeout immediately expires.
Steps to reproduce the error:

1. We launch a firmware that has SNTP and MQTT without an active Internet connection. MQTT reconnect time 15 seconds, keepalive 30 seconds. The system time starts counting from 0. The MQTT module is initialized and sets the keepalive time stamp from the system time (for example 1200). MQTT is unable to establish a connection and is trying to connect every 15 seconds.
2. We activate any Internet connection. SNTP performs time synchronization and the system time is unix-time (for example 1653291993). After that, MQTT successfully connects to the broker. The keepalive_process procedure immediately determines that the keepalive time has expired (1653291993 - 1200 = 1653290793, which is much more than 30000). The procedure sends a keepalive request. At the next iteration of the mqtt task, keepalive_process determines that the keepalive response has not arrived, since the broker does not have time to send the response. After that, the MQTT connection is terminated on the ESP32 side. After the reconnect timeout, the procedure is repeated and each attempt to connect to the MQTT broker fails with a keepalive response error.

Correction Suggestions:
1. Keepalive time stamp updates on MQTT connection event, not just MQTT initialization.
2. Using FreeRTOS counter instead of system time. In this case, SNTP will not affect reconnect and keepalive counters in the MQTT module.

Tested on eps-mqtt commit [64f88b4](https://github.com/espressif/esp-mqtt/tree/64f88b4412ea6649dbf207a07370c2617160d044)
Log:
```
E (00:00:22.024) esp-tls: couldn't get hostname for :22222222222222222-ats.iot.eu-west-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
E (00:00:22.026) esp-tls: Failed to open new connection
E (00:00:22.027) TRANSPORT_BASE: Failed to open a new connection
E (00:00:22.028) mqtt_client: Error transport connect
I (00:00:22.030) MQTT: DISCONNECTED
I (00:00:24.904) NET-MGR: Ethernet Link Up
I (00:00:24.905) NET-MGR: Ethernet HW Addr 7e:df:a1:03:67:85
I (00:00:25.524) NET-MGR: Ethernet got IP address
I (00:00:25.525) NET-MGR: ~~~~~~~~~~~~~~~~~~~~~~~~~
I (00:00:25.526) NET-MGR: ETH IP:   192.168.2.186
I (00:00:25.527) NET-MGR: ETH GW:   192.168.1.1
I (00:00:25.528) NET-MGR: ETH MASK: 255.255.252.0
I (00:00:25.529) NET-MGR: ~~~~~~~~~~~~~~~~~~~~~~~~~
I (00:00:25.530) NET-MGR: Ethernet connected!
I (00:00:25.531) esp_netif_handlers: eth ip: 192.168.2.186, mask: 255.255.252.0, gw: 192.168.1.1
I (00:00:38.794) MQTT: CONNECTED
E (00:00:39.065) mqtt_client: No PING_RESP, disconnected
I (00:00:39.071) MQTT: DISCONNECTED
I (08:07:41.554) MQTT: CONNECTED
E (08:07:41.825) mqtt_client: No PING_RESP, disconnected
I (08:07:41.831) MQTT: DISCONNECTED
I (08:07:58.503) MQTT: CONNECTED
E (08:07:58.780) mqtt_client: No PING_RESP, disconnected
I (08:07:58.787) MQTT: DISCONNECTED
```

ser90368141, 2022-05-23:
What else is seen. It can be seen from the logs above that the presence of SNTP is not necessary for the error to trigger.
It is enough that the time between the initialization of MQTT and the appearance of an Internet connection is greater than the MQTT keepalive timeout.

KonssnoK, 2022-05-23:
i think i had the same issue.
i solved by updating the keepalive timer on successful connection in the mqtt_client.

```
        client->event.event_id = MQTT_EVENT_CONNECTED;
        client->event.connected.session_present = session_present;
        client->state = MQTT_STATE_CONNECTED;
        esp_mqtt_dispatch_event(client);
        client->refresh_connection_tick = platform_tick_get_ms();
        client->keepalive_tick = platform_tick_get_ms();
```

last line

closed 2022-07-05 by david-cermak