espressif/esp-mqtt #229: A single Wifi forced disconnection blocks device from reconnecting forever.

KonssnoK, 2022-06-23:
Top of v4.4 branch 6c5fb29c2c513936ecefa7cdf49fb324e316546e

To replicate:
- connect to a phone hotspot
- switch off the hotspot
- switch back on the hotspot

```

I (1829590) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:11
I (09:39:51.464) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:1
E (09:39:51.464) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (09:39:51.480) MQTT_CLIENT: Poll read error: 119, aborting connection
I (09:39:51.488) mqtt_app: MQTT_EVENT_DISCONNECTED
E (1830621) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (09:39:53.798) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (09:39:56.129) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (09:39:58.461) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
E (1837585) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (1838895) wifi:new:<11,2>, old:<11,0>, ap:<11,2>, sta:<11,0>, prof:11
I (1838896) wifi:state: init -> auth (b0)
I (1838910) wifi:state: auth -> assoc (0)
I (1838947) wifi:state: assoc -> run (10)
I (1838991) wifi:connected with KI, aid = 1, channel 11, BW20, bssid = fa:70:51:63:4d:60
I (1838992) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (1838996) wifi:pm start, type: 0

I (1838997) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (1839006) wifi:BcnInt:102400, DTIM:2
I (09:40:00.877) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:fa:70:51:63:4d:60<ROOT>, ID:14:12:bf:4d:d9:a0
I (09:40:00.886) mesh_netif: Already wifi station, no need to do anything
I (09:40:00.894) mesh_main: <MESH_EVENT_TODS_REACHABLE>state:0
I (09:40:00.901) mesh_main: <MESH_EVENT_ROOT_ADDRESS>root address:7c:df:a1:e1:d3:5d
I (09:40:01.492) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (09:40:01.791) network: Network connected
I (09:40:01.791) esp_netif_handlers: sta ip: 192.168.180.12, mask: 255.255.255.0, gw: 192.168.180.241
I (09:40:01.792) network: IP: 192.168.180.12
I (09:40:01.800) network: Netmask: 255.255.255.0
I (09:40:01.806) network: Gateway: 192.168.180.241
I (09:40:01.827) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
W (1842993) wifi:<ba-add>idx:0 (ifx:0, fa:70:51:63:4d:60), tid:0, ssn:3, winSize:64
I (09:40:06.261) mqtt_app: MQTT_EVENT_CONNECTED
I (09:40:06.267) mqtt_app: sent publish returned msg_id=27106
I (09:40:06.359) mqtt_app: MQTT_EVENT_SUBSCRIBED, msg_id=34013
I (09:40:06.470) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=27106
E (09:40:07.490) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
I (09:40:07.490) mqtt_app: MQTT_EVENT_CONNECTED
I (09:40:07.491) mqtt_app: MQTT_EVENT_ERROR
E (09:40:07.498) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
I (09:40:07.509) mqtt_app: MQTT_EVENT_DISCONNECTED
E (09:40:07.512) MQTT_CLIENT: Client has not connected
I (09:40:07.518) MQTT_CLIENT: Client asked to disconnect
I (09:40:17.518) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (09:40:19.005) mqtt_app: MQTT_EVENT_CONNECTED
E (09:40:19.009) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
I (09:40:19.010) mqtt_app: MQTT_EVENT_ERROR
I (09:40:19.013) mqtt_app: sent publish returned msg_id=44280
E (09:40:19.014) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
I (09:40:19.031) mqtt_app: MQTT_EVENT_DISCONNECTED
```

euripedesrocha, 2022-06-27:
Hi @KonssnoK, could you share a small code example showing your network setup and MQTT configuration?

KonssnoK, 2022-06-28:
@euripedesrocha 

Network setup:
- ESP32S3
- Phone hotspot
- Internet

MQTT configuration:
[image]

Code Snippets are basically the ones of the DS example
```
    esp_mqtt_client_config_t mqtt_cfg = {
        .uri = server,
        .cert_pem = (const char*)server_crt,
        .client_cert_pem = (const char*)client_crt,
        .client_key_pem = NULL,
        .ds_data = ds_data,
        .username = user,
        .password = pass,
        .client_id = client_id
    };

    self->client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(self->client, ESP_EVENT_ANY_ID, mqtt_event_handler, self->client);
    // Starts MQTT
    esp_mqtt_client_start(self->client);
```
```
// NOTE: This function is running in the MQTT_TASK thread
static esp_err_t mqtt_event_handler_cb(esp_mqtt_event_handle_t event)
{
    switch (event->event_id) {
    case MQTT_EVENT_ERROR:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
        break;
    }
    case MQTT_EVENT_CONNECTED:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        self->connected = true;
        if (esp_mqtt_client_subscribe(self->client, mqtt_subscribe_topic_id, (int)MQTT_QOS_EXACTLY_ONCE) < 0) {
            // Disconnect to retry the subscribe after auto-reconnect timeout
            esp_mqtt_client_disconnect(self->client);
            self->connected = false;
        }

        if (self->connected) {
            announce_send_packet();
        }
        break;
    }
    case MQTT_EVENT_DISCONNECTED:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
        self->connected = false;
        break;
    }
    case MQTT_EVENT_SUBSCRIBED:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    }
    case MQTT_EVENT_UNSUBSCRIBED:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    }
    case MQTT_EVENT_PUBLISHED:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
        break;
    }
    case MQTT_EVENT_DATA:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_DATA");
        ESP_LOGI(TAG, "TOPIC=%.*s", event->topic_len, event->topic);

        processor_send_message(PROC_MSG_MQTT, (uint8_t*)event->data, event->data_len);
        break;
    }
    case MQTT_EVENT_BEFORE_CONNECT:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_BEFORE_CONNECT");
        break;
    }
    case MQTT_EVENT_DELETED:
    {
        ESP_LOGI(TAG, "MQTT_EVENT_DELETED");
        break;
    }
    default:
    {
        ESP_LOGI(TAG, "MQTT Other event id:%d", event->event_id);
        break;
    }
    }
    return ESP_OK;
}

```

euripedesrocha, 2022-06-28:
@KonssnoK sorry, by networking I meant the ESP32 connection configuration. I'm asking about this due to the mesh related events on your logs.