espressif/esp-mqtt #235: mqtt idf 4.4 azure iot hub

sl2121, 2022-09-19:
hi, i would like help on how connect to the azure iot hub withe "mqtt_client" library.

my configuration is the following :

static const unsigned  char DigiCertGlobalRootG2[] = (
"-----BEGIN CERTIFICATE-----\r\n"
"MIIDjjCCAnagAwIBAgIQAzrx5qcRqaC7KGSxHQn65TANBgkqhkiG9w0BAQsFADBh\r\n"
"MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\r\n"
"d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBH\r\n"
"MjAeFw0xMzA4MDExMjAwMDBaFw0zODAxMTUxMjAwMDBaMGExCzAJBgNVBAYTAlVT\r\n"
"MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j\r\n"
"b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IEcyMIIBIjANBgkqhkiG\r\n"
"9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuzfNNNx7a8myaJCtSnX/RrohCgiN9RlUyfuI\r\n"
"2/Ou8jqJkTx65qsGGmvPrC3oXgkkRLpimn7Wo6h+4FR1IAWsULecYxpsMNzaHxmx\r\n"
"1x7e/dfgy5SDN67sH0NO3Xss0r0upS/kqbitOtSZpLYl6ZtrAGCSYP9PIUkY92eQ\r\n"
"q2EGnI/yuum06ZIya7XzV+hdG82MHauVBJVJ8zUtluNJbd134/tJS7SsVQepj5Wz\r\n"
"tCO7TG1F8PapspUwtP1MVYwnSlcUfIKdzXOS0xZKBgyMUNGPHgm+F6HmIcr9g+UQ\r\n"
"vIOlCsRnKPZzFBQ9RnbDhxSJITRNrw9FDKZJobq7nMWxM4MphQIDAQABo0IwQDAP\r\n"
"BgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBhjAdBgNVHQ4EFgQUTiJUIBiV\r\n"
"5uNu5g/6+rkS7QYXjzkwDQYJKoZIhvcNAQELBQADggEBAGBnKJRvDkhj6zHd6mcY\r\n"
"1Yl9PMWLSn/pvtsrF9+wX3N3KjITOYFnQoQj8kVnNeyIv/iPsGEMNKSuIEyExtv4\r\n"
"NeF22d+mQrvHRAiGfzZ0JFrabA0UWTW98kndth/Jsw1HKj2ZL7tcu7XUIOGZX1NG\r\n"
"Fdtom/DzMNU+MeKNhJ7jitralj41E6Vf8PlwUHBHQRFXGU7Aj64GxJUTFy8bJZ91\r\n"
"8rGOmaFvE7FBcf6IKshPECBV1/MUReXgRPTqh5Uykw7+U0b6LJ3/iyK5S9kJRaTe\r\n"
"pLiaWN0bfVKfjllDiIGknibVb63dDcY3fe0Dkhvld1927jyNxF1WW6LZZm6zNTfl\r\n"
"MrY=\r\n"
"-----END CERTIFICATE-----\r\n"
);

const esp_mqtt_client_config_t mqtt_cfg = {

			.uri = "mqtts://myhubname.azure-devices.net:8883",
			.client_id = "device",
	                .username = "myhubname.azure-devices.net/device/?api-version=2020-09-30",
			.password = MQTT_PASSWD,//(SAS Token)
			.cert_pem =(const char *)DigiCertGlobalRootG2,
			//.cert_len = sizeof(certificates),
			//.port = 8883,
			//.disable_auto_reconnect = true,
			.lwt_qos = 1,
			//.transport = MQTT_TRANSPORT_OVER_SSL,
			.protocol_ver = MQTT_PROTOCOL_V_3_1_1,
			//.use_secure_element = true,
			.lwt_topic = "/lwt",
			.lwt_msg = "offline",
			.lwt_retain = 0

	};
esp_err_t err;
esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
err = esp_mqtt_client_start(client);
ESP_LOGI (TAG, "Client connect. Error = %d %s", err, esp_err_to_name (err));

With these settings i can't connect.
Thanks

euripedesrocha, 2022-09-21:
Hi @sl2121, could you share the logs for the failed connection?

sl2121, 2022-09-22:
I have started from the sntp example and i continued with the mqtt part.
 What is the meaning of : esp-tls-mbedtls: Certificate verified.
Is that the esp has verified that the certificate is correct ?
Thanks.

[0;32mI (17122) MQTTS_EXAMPLE: Other event id:7[0m
[0;31mE (23882) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7080[0m
[0;32mI (23892) esp-tls-mbedtls: Certificate verified.[0m
[0;31mE (23892) esp-tls: Failed to open new connection[0m
[0;31mE (23892) TRANSPORT_BASE: Failed to open a new connection[0m
[0;31mE (23902) MQTT_CLIENT: Error transport connect[0m
[0;32mI (23902) MQTTS_EXAMPLE: MQTT_EVENT_ERROR[0m
[0;32mI (23912) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x801a[0m
[0;32mI (23922) MQTTS_EXAMPLE: Last tls stack error number: 0x7080[0m
[0;32mI (23922) MQTTS_EXAMPLE: Last captured errno : 0 (Success)[0m
[0;32mI (23932) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED[0m

sl2121, 2022-09-22:
hi i have loaded the ESP-MQTT SSL Sample application.

static void mqtt_app_start(void)
{
    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = CONFIG_BROKER_URI,
        .cert_pem = (const char *)mqtt_eclipseprojects_io_pem_start,
    };
mqtt_eclipseprojects_io_pem_start inside here i have put the  DigiCertGlobalRootG2[] "MIIDjjCCAn...MrY"

and i see :

[0;32mI (6592) example_connect: Got IPv4 event: Interface "example_connect: sta" address: *********
[0;32mI (6602) example_connect: Connected to example_connect: sta[0m
[0;32mI (6612) example_connect: - IPv4 address: ***********
[0;32mI (6612) example_connect: - IPv6 address: fe80:0000:0000:0000:4291:51ff:feb2:8e74, type: ESP_IP6_ADDR_IS_LINK_LOCAL[0m
[0;32mI (6622) MQTTS_EXAMPLE: [APP] Free memory: 234100 bytes[0m
[0;32mI (6632) MQTTS_EXAMPLE: Other event id:7[0m
[0;32mI (15212) MQTTS_EXAMPLE: MQTT_EVENT_CONNECTED[0m
[0;32mI (15222) MQTTS_EXAMPLE: sent subscribe successful, msg_id=53250[0m
[0;32mI (15222) MQTTS_EXAMPLE: sent subscribe successful, msg_id=60112[0m
[0;32mI (15222) MQTTS_EXAMPLE: sent unsubscribe successful, msg_id=18944[0m
[0;32mI (15422) MQTTS_EXAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=53250[0m
[0;32mI (15432) MQTTS_EXAMPLE: sent publish successful, msg_id=0[0m
[0;32mI (15472) MQTTS_EXAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=60112[0m
[0;32mI (15482) MQTTS_EXAMPLE: sent publish successful, msg_id=0[0m
[0;32mI (15632) MQTTS_EXAMPLE: MQTT_EVENT_DATA[0m
TOPIC=/topic/qos1 DATA=data_3

sl2121, 2022-09-22:
After in the 
static void mqtt_app_start(void)
{
    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://miothub.azure-devices.net:8883",
        .cert_pem = (const char *)mqtt_eclipseprojects_io_pem_start,
    };

i see 

[0;32mI (6612) example_connect: - IPv4 address: ***********
[0;32mI (6612) example_connect: - IPv6 address: *******************, type: ESP_IP6_ADDR_IS_LINK_LOCAL[0m
[0;32mI (6622) MQTTS_EXAMPLE: [APP] Free memory: 234100 bytes[0m
[0;32mI (6632) MQTTS_EXAMPLE: Other event id:7[0m
[0;33mW (23892) esp-tls: Failed to open new connection in specified timeout[0m
[0;31mE (23892) TRANSPORT_BASE: Failed to open a new connection[0m
[0;31mE (23892) MQTT_CLIENT: Error transport connect[0m
[0;32mI (23892) MQTTS_EXAMPLE: MQTT_EVENT_ERROR[0m
[0;32mI (23902) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x8006[0m
[0;32mI (23912) MQTTS_EXAMPLE: Last tls stack error number: 0x0[0m
[0;32mI (23912) MQTTS_EXAMPLE: Last captured errno : 0 (Success)[0m
[0;32mI (23922) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED[0m
[0;32mI (38922) MQTTS_EXAMPLE: Other event id:7[0m
[0;33mW (55842) esp-tls: Failed to open new connection in specified timeout[0m
[0;31mE (55842) TRANSPORT_BASE: Failed to open a new connection[0m
[0;31mE (55842) MQTT_CLIENT: Error transport connect[0m
[0;32mI (55842) MQTTS_EXAMPLE: MQTT_EVENT_ERROR[0m
[0;32mI (55852) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x8006[0m
[0;32mI (55852) MQTTS_EXAMPLE: Last tls stack error number: 0x0[0m
[0;32mI (55862) MQTTS_EXAMPLE: Last captured errno : 0 (Success)[0m
[0;32mI (55872) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED[0m
[0;32mI (70872) MQTTS_EXAMPLE: Other event id:7[0m

Thanks.

closed 2022-10-28 by Alvin1Zhang