espressif/esp-mqtt #237: Add support for getting unsent messages out of mqtt outbox

bbinet, 2022-09-21:
I'm using esp-mqtt in a battery powered project, so to save energy I'm deep sleeping most of the time.

I send sensor data from time to time through mqtt, but if it takes too much time, I would like to return to deep sleep, and save the data which has not been actually sent (ack has not been received) to flash memory (NVS), so that I can try again the next time I wake up.

I couldn't find any API to get unsent data out of mqtt outbox.
Can you think of any way to do it?

bbinet, 2022-09-21:
Also when `MQTT_EVENT_DELETED` event is triggered, it would be helpful to be able to retrieve the data which has been deleted in case we want to do anything with it.

euripedesrocha, 2022-09-21:
Hi @bbinet, in your case, I would suggest taking the route of writing your own outbox.

Alvin1Zhang, 2022-10-24:
Thanks for reporting, would close due to short of feedback, feel free to reopen.

closed 2022-10-24 by Alvin1Zhang