espressif/esp-mqtt #238: MQTT client resends message with qos = 1

LisaTamanti1965, 2022-10-07:
Hi

I'm started to work with ESP32 configured as MQTT client.
I noted that when I send a message with qos = 1 (in the example below, msg with id = m23462) the device receives more than one PUBACK...I guess that is due to dalay in receiving PUBACK...so the message is repeated a number of times I received PUBACK.
No LOG when the message is sent again?
how many time to wait PUBACK before to resend message? it's possible to change this time?

I (54957) CLOUD_COMM: Sending 0 from CH 1 (m23462): {"cmd":0,"pyl":{"type":"A5","id":"7873481CSC02191210","fw":"2.3","hw":"1.0","com":3,"ip":"100.106.96.16","netmask":"255.255.255.255","gateway":"10.64.64.64","mac":"246F284DE56C","imsi":"901405100299026","imei":"866425033022395","iccid":"89882806660002990267","rev":"","oper":"vodafone IT 1nce.net","oper_type":1,"rssi":14,"simstatus":1,"simexp":871660947,"idsimservice":"","wl":2},"ts":"2ad31357","tsid":"c975"}
V (55341) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (56344) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (57348) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (58352) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (58492) MQTT_CLIENT: mqtt_message_receive: first byte: 0x40
D (58495) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (58497) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (58510) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (58511) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (58523) MQTT_CLIENT: msg_type=4, msg_id=23462
D (58524) MQTT_CLIENT: pending_id=23462, pending_msg_count = 1
D (58526) OUTBOX: DELETED msgid=23462, msg_type=3, remain size=0
D (58537) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBACK, finish QoS1 publish
D (58539) event: running post MQTT_EVENTS:5 with handler 0x400f5778 and context 0x3ffe3908 on loop 0x3ffe37bc
I (58552) MQTT_Wrapper: MQTT_EVENT_PUBACK, msg_id=23462
D (58669) MQTT_CLIENT: mqtt_enqueue id: 43371, type=3 successful
D (58671) OUTBOX: ENQUEUE msgid=43371, msg_type=3, len=192, size=192
I (58674) CLOUD_COMM: Sending 13 from CH 1 (m43371): {"cmd":13,"pyl":{"type":1,"board":"E1RD","cfg":"A1400RD","fw":"4.1","hw":"4.7","accessIdx":1,"auxt":[],"auxfw":[]},"ts":"2ad31357","tsid":"c976"}
V (59564) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (60567) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (61571) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (62574) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (63211) MQTT_CLIENT: mqtt_message_receive: first byte: 0x40
D (63213) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (63215) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (63228) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (63229) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (63241) MQTT_CLIENT: msg_type=4, msg_id=23462
D (63243) MQTT_CLIENT: pending_id=43371, pending_msg_count = 1
D (63245) MQTT_CLIENT: mqtt_message_receive: first byte: 0x40
D (63257) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (63259) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (63272) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (63273) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (63285) MQTT_CLIENT: msg_type=4, msg_id=23462
D (63286) MQTT_CLIENT: pending_id=43371, pending_msg_count = 1
V (64288) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (65292) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (66296) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (67300) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (67594) MQTT_CLIENT: mqtt_message_receive: first byte: 0x40
D (67597) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (67599) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (67613) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (67614) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (67626) MQTT_CLIENT: msg_type=4, msg_id=23462
D (67627) MQTT_CLIENT: pending_id=43371, pending_msg_count = 1
D (67629) MQTT_CLIENT: mqtt_message_receive: first byte: 0x40
D (67641) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (67644) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (67657) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (67658) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (67660) MQTT_CLIENT: msg_type=4, msg_id=43371
D (67671) MQTT_CLIENT: pending_id=43371, pending_msg_count = 1

euripedesrocha, 2022-10-07:
Hi @LisaTamanti1965, you are correct. If the PUBACK is delayed, the message is transmitted again.

 We recently added the log for duplicated message sent, so it should be available depending on the version you are using. 

It's possible to control the timeout through the `message_retransmit_timeout` field in the configuration. The behavior is the same for both QoS 1 and 2 messages.

LisaTamanti1965, 2022-10-07:
I'm using ESP-IDF 4.4.1

LisaTamanti1965, 2022-10-07:
Sometimes I have this type of LOG (see below) what could be the cause?
Connection by GSM is closed or other?

I (1253690) CLOUD_COMM: Sending 1 from CH 1 (m0): {"cmd":1,"pyl":{"state":7,"modfun":10,"manut":0,"uid":"","tot_c":721,"rel_c":0,"sf":0,"maxsf":0,"log":0,"adv":0,"out":[1,1]},"ts":"2ad31806","tsid":"1315ce"}
V (1254294) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
E (1255217) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (1255220) MQTT_CLIENT: Poll read error: 119, aborting connection

LisaTamanti1965, 2022-10-10:
I have an other question about MQTT....if the device doesn't receive PUBACK.....continues to send messages? How many messages are sent before to stop? Or how many time remain in this status (resending msg)?
Is it possible to set how many resend  or how many time?

euripedesrocha, 2022-10-11:
Hi @LisaTamanti1965,
in the shared log there was a failure in receiving the data and the client will abort connection and reconnect, depending on your configuration.

About the messages that don't have a PUBACK there will be as many retries as possible until `CONFIG_MQTT_OUTBOX_EXPIRED_TIMEOUT_MS`  when the message will be deleted. It's possible to have an event if you configure the client to notify when deleting messages.

LisaTamanti1965, 2022-10-12:
Thank you very much for your help!!!!

closed 2022-10-12 by LisaTamanti1965