espressif/esp-mqtt #245: Make `MQTT_POLL_READ_TIMEOUT_MS` configurable through `menuconfig/sdkconfig`

ItayDumayVayyar, 2023-01-20:
Hey, I have implemented my own [`MQTT` outbox](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mqtt-custom-outbox) for storing persistent messages in NVS. But I encountered a problem. 
In `esp_mqtt_task()` the main logic is to ask the outbox for an item, transmit it, and then poll read for a potential ACK.  
https://github.com/espressif/esp-mqtt/blob/bb9c8af9d552b608dd3aabf9617bde757a538ebe/mqtt_client.c#L1562-L1568
As we can see `esp_transport_poll_read()` is used, **where `MQTT_POLL_READ_TIMEOUT_MS` is a fixed and hard coded value**. This implies that `esp_mqtt_client_enqueue()` can not be used as the rate would be very slow - it creates a situation where there is a clear dependency between messages, the next message will be requested from the outbox only if the timeout (1s currently) has passed or an ACK has been received.

euripedesrocha, 2023-02-02:
Hi @ItayDumayVayyar, thanks for the suggestion. I'm working on this feature, it should be available soon.

ItayDumayVayyar, 2023-02-02:
> Hi @ItayDumayVayyar, thanks for the suggestion. I'm working on this feature, it should be available soon.

Great, thanks! Can you estimate in what ESP-IDF version this change will be included?

closed 2023-02-02 by david-cermak