espressif/esp-mqtt #247: Publish data leaking to topic

DEADSEC-SECURITY, 2023-02-01:


DEADSEC-SECURITY, 2023-02-01:
Also after logging the contents of data I get the correct data if the data is "High" but if I send "Low" the log outputs `Lowâ””` or `Lowh`, basically Low with some random chars. (Maybe uninitialized memory?)

DEADSEC-SECURITY, 2023-02-01:
Some EMQX logs in case it helps visualize the data being sent:
```
2023-02-01T12:45:08+00:00 [PUBLISH] 3s31eT307G2DbsDJXJjKCp@10.1.10.7:33767 msg: publish_to, topic: homeassistant/select/grandmother_pill_dispenser_volume/set, payload: Low
2023-02-01T12:48:49+00:00 [SUBSCRIBE] Grandmother Pill Dispenser@10.4.81.207:58148 msg: subscribe, sub_id: Grandmother Pill Dispenser, sub_opts: [nl: 0, qos: 0, rap: 0, rh: 0, sub_props: []], topic: homeassistant/select/grandmother_pill_dispenser_volume/set
2023-02-01T12:48:53+00:00 [SUBSCRIBE] Grandmother Pill Dispenser@10.4.81.207:55043 msg: subscribe, sub_id: Grandmother Pill Dispenser, sub_opts: [nl: 0, qos: 0, rap: 0, rh: 0, sub_props: []], topic: homeassistant/select/grandmother_pill_dispenser_volume/set
```

ESP-YJM, 2023-02-03:
@DEADSEC-SECURITY I think you can't use `%s` to print `msg_topic`, you can use following code to print. `msg_topic` is not end of `\0`, so you can't use strlen to calcaute its length.

```
ESP_LOGI(TAG, "Topic is %.*s", msg_topic_len, msg_topic);
```

euripedesrocha, 2023-02-03:
Hi @DEADSEC-SECURITY as pointed by @ESP-YJM `msg_topic` isn't null terminated and part of the same buffer. You need to take the size in consideration when handling them, for data if the size is bigger than the internal buffer multiple events are dispatched.

DEADSEC-SECURITY, 2023-02-06:
I didnt have time to test this out cuz another bug appeared but once thats fixed ill check this out. Thanks @euripedesrocha and @ESP-YJM

closed 2023-02-10 by DEADSEC-SECURITY