espressif/esp-mqtt #251: How can I detect Expired certificates

LisaTamanti1965, 2023-02-17:
When I try a MQTT connection how can I detect an expired certificate from values in

event->error_handle->error_type, 
event->error_handle->connect_return_code, 
event->error_handle->esp_tls_cert_verify_flags

reported in MQTT_EVENT_ERROR ?

I haven't expired certificate to do some tests....

david-cermak, 2023-02-23:
Checking for certificate validity is disabled by default in IDF/mbedTLS. You can enable in menuconfig setting [`CONFIG_MBEDTLS_HAVE_TIME_DATE`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-mbedtls-have-time-date) to `true`.

If enabled and expired you should see the first bit *set* in the flag variable:
```cpp
#define MBEDTLS_X509_BADCERT_EXPIRED             0x01  /**< The certificate validity has expired. */
```
set in the `event->error_handle->esp_tls_cert_verify_flags`

and the error code from mbedTLS would be
```cpp
#define MBEDTLS_ERR_X509_CERT_VERIFY_FAILED               -0x2700
```

so the MQTT library error log should be something like:
```
D (19697) esp-tls: handshake in progress...
E (19707) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (19717) esp-tls-mbedtls: (FFFFD900): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
I (19717) esp-tls-mbedtls: Failed to verify peer certificate!
E (19727) esp-tls: Failed to open new connection
E (19727) transport_base: Failed to open a new connection
E (19737) mqtt_client: Error transport connect
D (19737) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=0
I (19747) MQTT_EXAMPLE: MQTT_EVENT_ERROR
E (19757) MQTT_EXAMPLE: Last error reported from esp-tls: 0x801a
E (19757) MQTT_EXAMPLE: Last error reported from tls stack: 0x2700
W (19767) MQTT_EXAMPLE: esp_tls_cert_verify_flags=9
I (19777) MQTT_EXAMPLE: Last errno string (Success)
```

(Note that SNTP needs to be enabled and started for the cert expiration check to work)

---
I'd suggest generating and using self-signed certificates in the local network to play with it and test how it works (you can use server side verification, or client side, or mutual).

LisaTamanti1965, 2023-02-28:
Thank you very much!!! Exactly what I need!

david-cermak, 2023-03-27:
Good to hear that was helpful, closing

closed 2023-03-27 by david-cermak