espressif/esp-mqtt #252: ESP-IDF won't connect to MQTT broker local gateway IP

SandorMatus, 2023-02-23:
Hi
I have a very peculiar issue. I am using ESP32 with IDF and trying to send some data to my MQTT broker running on my router through WiFi. The thing is that ESP does not want to connect to local gateway IP on the WiFi interface. The connection to the broker works on my windows PC connected to the same WiFi. The only time I can get the ESP to connect to my broker is when I plug in some other interface like eth0 to internet and set DHCP and on ESP I set the IP of the eth0 interface of my router. 
I cannot get to the bottom of this, I have checked LWIP logs and it seems the ESP drops the packet entirely and it does not get to the router at all. 

Br.
Matt

david-cermak, 2023-02-24:
Hi @SandorMatus 

Could you please describe your network configuration in more detail? Also the network settings on ESP32? (Is it the default WiFi station with DHCP client running)?
Does the ESP32 get an IP address? Could you ping the WiFI access point? Could you ping the broker? (assume it's: yes, yes, no) Trying to access the broker via it's address or name?
Could you please check the outgoing traffic with wireshark?
Also, it would be helpful if you post the logs from ESP32.

SandorMatus, 2023-02-24:
Hi David
So the network configuration is a bit strange, I am trying to figure out how it works myself but I am using ESP mesh to interconnect multiple devices between each other to create route to my main router with MQTT broker. I will attach the logs later when I get to my office. Meanwhile I figured out that changing DHCP pools and reseting leases sometimes changes the behaviour of the issue. This could be an issue on the router side but it would not explain why it works from my Windows and not from an ESP. I have checked the LWIp logs again and it tries to connect 54 but it fails with error code -5. Meanwhile checking the wlan interface on my router I see no attempt to handshake (using tcpdump). For now I have to set a static IP on my eth0 interface and set the IP on the ESP (this resolves the issue but still does not make sense why I need it like that). 
I will post logs and everything later this afternoon so you can have a look yourself.

Br.
Matt

SandorMatus, 2023-02-24:
Hi David
So upon further investigation it seems that LWip is using APs IP address as source (using local ip as mqtt broker) and this is creating a routing error, see attached logs. 
[image]
And of course it does not show up in router because the src IP is wrong.
[image]
Is there any way to change the netif to take the STA IP instead of AP ? I think this is probably no longer a MQTT issue so if you want we can close this issue.

Br.
Matt

SandorMatus, 2023-02-24:
Ok so this had nothing to do with anything and at the end it was a DNS issue.
I had overlaping networks :). Now I just use static DNS.

closed 2023-02-24 by SandorMatus