espressif/esp-mqtt #253: Init multiple network components in separate network interfaces

cinargursoy, 2023-02-23:
Hello,

I'm writing a  wrapper library over ETH, GSM & Wi-Fi connectivity. But my goal is to handle multiple Network Interfaces simultaneously, and be able to seperate the Network Interfaces of different tasks. 

 For example, what do you recommend for a scenario like this:

> There is a project with two separated ETH chips.  The client code wants to run RestServer and MQTT(or more) tasks trough separate networks simultaneously. 

I know that it's possible to create multiple `netif`'s. For libs like `esp-modbus`, `esp_http_client`etc. it's possible to initialize them via `netif`. If every lib were able to init via `netif`, that would not be a problem for me.
But  components  like `mqtt` are using `tcp-transport`, and it's impossible to initialize the component by giving `netif`. I want to be able to route each component's network interface. 

I wathced the presentation [EDC22 Day 1 Talk 11](https://www.youtube.com/watch?v=tCpxJNnFfMM). It's mentioned that, multiple interfaces may be handled by priorities of connection peripheral. But, how can I use that for my use case or any other recomendation would be great.

david-cermak, 2023-02-23:
Hello @cinargursoy 

The network libraries that use the `tcp-transport` have to *bind* the underlying socket/transport to the specified interface. It's unfortunately not possible to configure it independently, just setting:

[esp_transport_tcp_set_interface_name()](https://github.com/espressif/esp-idf/blob/778aeae99ebe7d0fe437ae4aa73a9f74c5a83668/components/tcp_transport/include/esp_transport_tcp.h#L34)

but needs to be specifically added to the config struct of the actual network library. MQTT client doesn't support this config option, yet.

We can add this option, the same way it's supported in [esp_http_client](https://github.com/espressif/esp-idf/blob/778aeae99ebe7d0fe437ae4aa73a9f74c5a83668/components/esp_http_client/include/esp_http_client.h#L141)

cinargursoy, 2023-02-23:
Hello @david-cermak 

Thanks for your reply. 

> We can add this option, the same way it's supported in [esp_http_client](https://github.com/espressif/esp-idf/blob/778aeae99ebe7d0fe437ae4aa73a9f74c5a83668/components/esp_http_client/include/esp_http_client.h#L141)

Great! That would solve the issue. What would be the predicted date for this feature?

closed 2023-06-14 by david-cermak