espressif/esp-mqtt #261: Possible bug on mqtt5 client when QoS>0

jmpmscorp, 2023-06-16:
### Enviroment

- Windows 10
- esp-wrover-kit (esp32)
- esp-idf v5.0.2 release

### Description

Reason code on messages with QoS1 or QoS2 don't return 0x00 when brokers don't add reason code in PUBACK, PUBREL, PUBREC and PUBCOMP packets and its remaining length are exactly 2.

Tested with:

<details>

<summary>
Code
</summary>
 
```c

#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "esp_check.h"
#include "esp_system.h"
#include "esp_event.h"

#include "protocol_examples_common.h"

#include "nvs_flash.h"
#include "nvs.h"
#include "mqtt_client.h"

#define MQTT_BROKER_MOSQUITTO 1
#define MQTT_BROKER_HIVEMQ 2
#define MQTT_BROKER_EMQX 3
#define MQTT_BROKER_ECLIPSE_PROJECT 4

#define MQTT_BROKER_USED 4
#define MQTT_QOS_USED 2

#define MQTT_BROKER_MOSQUITTO 1
#define MQTT_BROKER_HIVEMQ 2
#define MQTT_BROKER_EMQX 3
#define MQTT_BROKER_ECLIPSE_PROJECT 4

#define MQTT_BROKER_USED 3
#define MQTT_QOS_USED 2

#if MQTT_BROKER_USED == MQTT_BROKER_MOSQUITTO
#define MQTT_BROKER_URL "mqtt://test.mosquitto.org"
#define MQTT_BROKER_PORT 1883

#elif MQTT_BROKER_USED == MQTT_BROKER_HIVEMQ
#define MQTT_BROKER_URL "mqtt://broker.hivemq.com"
#define MQTT_BROKER_PORT 1883

#elif MQTT_BROKER_USED == MQTT_BROKER_EMQX
#define MQTT_BROKER_URL "mqtt://broker.emqx.io"
#define MQTT_BROKER_PORT 1883

#elif MQTT_BROKER_USED == MQTT_BROKER_ECLIPSE_PROJECT
#define MQTT_BROKER_URL "mqtt://mqtt.eclipseprojects.io"
#define MQTT_BROKER_PORT 1883
[... 177 lines cut ...]
[image]
</details>

Yellow log message is the packet received by the broker. I add this line on ``mqtt5_msg_get_reason_code`` function. Same log with same local broker docker versions .

<details>

<summary>
Log on emqx free online test broker. 
</summary>

[image]

</details>

With EMQX broker, message error `` E (29216) mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ack`` is not shown.

In OASIS mqtt specs in PUBACK, PUBREL, PUBREC and PUBCOMP packets we can find this: 
[image]

[image]

It seems that mosquitto, hivemq and eclipse broker tested don't append reason code when success.

As can be checked in logs, when broker doesn't append reason code (yellow lines), returned value is -1 but, by specs, it should be 0x00. I suppose that this -1 returned value is the cause that messages waiting to be confirmed by PUBxxx packets show  `` E (29216) mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ack`` error.

ESP-YJM, 2023-10-18:
@jmpmscorp Thanks for reporting it. We will fix it.

AxelLin, 2024-08-07:
> @jmpmscorp Thanks for reporting it. We will fix it.

@ESP-YJM
There is no update for 10 Months? How is the progress for this fix?

ESP-YJM, 2024-08-07:
@jmpmscorp Could you please use the latest MQTT components, maybe use IDF master branch to test the issue ` E (29216) mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ac`. And for `mqtt5_msg_get_reason_code ` return -1, the reason code is 0x0, it not affect the code logic and only a print issue. Sorry that i forget to fix it. Thanks for reminding @AxelLin

jmpmscorp, 2024-08-22:
Sorry for delay ESP-YJM, holiday time.

I have tested again with IDF 5.3 release version, QoS2 and no more E (29216) ```mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ack```.

Reason code -1 still printed. I know it will not affect but, maybe, in my opinion, it would be a good idea return another value than -1 that could be interpreted as unknown error and, certainly, it's not an error because it must be equal than 0x00.

Thanks for your work.

closed 2025-03-03 by euripedesrocha