espressif/esp-mqtt #263: Qos 2 wrong implementation

jyangav, 2023-07-10:
Read through the mqtt_client.c implementation, 
For publishing a qos2 message, seems there is no retransmission of the outbox message when its pending state is  TRANSMITTED. If send PUBREL or receive PUBCOMP failed, QOS 2 cannot guarantee message transmitted exactly once.

// resend all non-transmitted messages first
            outbox_item_handle_t item = outbox_dequeue(client->outbox, QUEUED, NULL);
            if (item) {
                if (mqtt_resend_queued(client, item) == ESP_OK) {
                    outbox_set_pending(client->outbox, client->mqtt_state.pending_msg_id, TRANSMITTED);
                }
                // resend other "transmitted" messages after 1s
            } else if (has_timed_out(last_retransmit, client->config->message_retransmit_timeout)) {
                last_retransmit = platform_tick_get_ms();
                item = outbox_dequeue(client->outbox, TRANSMITTED, &msg_tick);
                if (item && (last_retransmit - msg_tick > client->config->message_retransmit_timeout))  {
                    mqtt_resend_queued(client, item);
                }
            }

euripedesrocha, 2023-07-24:
Hi @jyangav, thanks for reporting. 

I don't understand the issue. The code you are showing will do a retransmission of the QoS 2 (and one) in the case that the follow-up messages are not received prior to the retransmission timeout.

jyangav, 2023-07-24:
# ESP-MQTT Configurations
#
CONFIG_MQTT_PROTOCOL_311=y
CONFIG_MQTT_TRANSPORT_SSL=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET_SECURE=y
# CONFIG_MQTT_MSG_ID_INCREMENTAL is not set
# CONFIG_MQTT_SKIP_PUBLISH_IF_DISCONNECTED is not set
CONFIG_MQTT_REPORT_DELETED_MESSAGES=y
CONFIG_MQTT_USE_CUSTOM_CONFIG=y
CONFIG_MQTT_TCP_DEFAULT_PORT=1883
CONFIG_MQTT_SSL_DEFAULT_PORT=8883
CONFIG_MQTT_WS_DEFAULT_PORT=80
CONFIG_MQTT_WSS_DEFAULT_PORT=443
CONFIG_MQTT_BUFFER_SIZE=1024
CONFIG_MQTT_TASK_STACK_SIZE=6144
# CONFIG_MQTT_DISABLE_API_LOCKS is not set
CONFIG_MQTT_TASK_PRIORITY=5
# CONFIG_MQTT_TASK_CORE_SELECTION_ENABLED is not set
# CONFIG_MQTT_CUSTOM_OUTBOX is not set
CONFIG_MQTT_OUTBOX_EXPIRED_TIMEOUT_MS=36000000
# end of ESP-MQTT Configurations

framework-espidf @ 3.40402.0 (4.4.2)

I am using PlatfromIO to do the compiling. I write a simple program to publish qos 2 message, it will start to publish second message only if publish event triggered. At the meantime, I randomly cut the wifi to replicate the worst network environment, it might stop publishing somehow. Seems the problem was what I mentioned above. The purpose is to make sure qos 2 won't lose any message even in the extremally bad network environment.
Sorry for anything unclear, it is first time I post the issue in github. Thx.

euripedesrocha, 2023-09-20:
@jyangav from the described scenario, you need to pay attention to the timeouts set for retransmission and expiration of messages. 
I suggest that you set [the client to report deleted messages](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mqtt-report-deleted-messages) and handle the worst case scenario within that event handler. 

I'll close this issue since it's related to project configuration and some scenarios where the library already has a mechanism to deal with. Feel free to reopen, adding more information, if you think I'm missing something.

closed 2023-09-20 by euripedesrocha