espressif/esp-mqtt #267: esp_mqtt_set_config ignores buffer_size

MathewHDYT, 2023-09-13:
The `esp_mqtt_set_config` does not actually increase or decrease the buffer size accordingly in the client configuration, meaning when changing the buffer size and calling `esp_mqtt_set_config` nothing changes the buffersize is still the initial value.

Instead to actually change the buffer size the complete `esp_mqtt` client has to be reinitalized.

Is this intended behaviour, meaning the buffer size can not be changed anymore after having initialised a `MQTT client`. Then it would be nice if this could be documented in the documentation.

If it is not intended, then I would appreciate that feature, because I would like to temporarily while the `OTA update` is ongoing increase the buffer size to receive the payload packets and then once it has finished decrease the buffer again, because I normally never receive packets that big (`4096 bytes`). If I can't do that with `esp_mqtt_set_config` then the buffer size always has to be 4KB which is a rather large chunk to create on the heap.

The missing update of the `bufffer_size` parameter can be found in the [`MQTT client`](https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L369) implementation.

euripedesrocha, 2023-09-14:
Hi @MathewHDYT, thanks for reporting the issue. For some configurations, it's required to restart the client, but it's not the case for the buffer, It's indeed a bug in the library.

MathewHDYT, 2023-09-14:
@euripedesrocha Thanks that is what I thought because increasing an internal buffer shouldn't have any dependency on the currently ongoing connection, the only thing that could cause problems if there is currently data in the buffer and the buffer size is made significantly smaller.

closed 2023-10-25 by david-cermak