espressif/esp-mqtt #268: AUTH command not supported

papadeltasierra, 2023-12-20:
MQTT v5 has an AUTH command that allows reauthentication without losing the connection.  This is important where authentication is using short-lived tokens as it allows a new token to be presented before the old token expires.  I don't see any support for this in this library and it would be good to have so that we avoid the old "disconnect, reconnect" juggling.

danielmeza, 2026-11-09:
We also need MQTT 5 enhanced authentication (spec 4.12), starting with the CONNECT side, which re-authentication builds on.

**Use case:** devices connect with mutual TLS and present a short signed token at CONNECT (Authentication Method + Authentication Data). A broker plugin verifies it once per connection. Mosquitto passes both fields to plugins (`MOSQ_EVT_EXT_AUTH_START`), but does not expose CONNECT user properties, so user properties are not an alternative.

**Proposed changes:**
1. `esp_mqtt5_connection_property_config_t`: add `authentication_method`, `authentication_data` and `authentication_data_len`, encoded in CONNECT as properties 0x15 and 0x16.
2. CONNACK: expose the returned Authentication Method/Data in the connack properties, and report reason codes 0x8C (Bad authentication method) and 0x87 (Not authorized).
3. AUTH packet: a new `MQTT_EVENT_AUTH` with method and data for an incoming AUTH (0x18 Continue authentication), and `esp_mqtt5_client_send_auth(client, reason_code, data, len)` to answer it or to start re-authentication (0x19). This covers the original request of this issue.

I can contribute a PR for 1 and 2, and for 3 if the API looks right to you. Is this a direction you would accept, or is there work in progress?

euripedesrocha, 2026-11-09:
@danielmeza Thanks for the added use case, we are working to add the advanced authentication. The message part is ready and we are working on the user facing API.

danielmeza, 2026-11-09:
Thanks @euripedesrocha, good to hear. We'll hold off on our own PR to avoid duplicating your work.

We can test it as soon as there is a branch: ESP32-S3 devices on mutual TLS against Mosquitto 2.0, with a plugin that checks the Authentication Method/Data at CONNECT (`MOSQ_EVT_EXT_AUTH_START`). Is there a branch we can follow, or a rough timeline?