espressif/esp-mqtt #274: MQTT data persistence (QOS 1 and 2)

Teesmo, 2024-04-15:
From the documentation,and after perusing through some of code, I gather that MQTT enqueues QOS level 1 and 2 messages (regardless of whether the persistence handler is enabled or not). My question is, where is this persisted data stored? Is it stored in RAM or in Flash? Will it exist (still be enqueued) if my device powers down?

euripedesrocha, 2024-04-16:
Data is stored in RAM, if you want to have some persistence to flash you can implement a custom outbox. We have [an example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/custom_outbox) showing usage of C++ allocators. You can use it to see the changes needed in the configuration and how to set the build.

closed 2024-05-23 by euripedesrocha

Teesmo, 2024-06-06:
@euripedesrocha Would you advise I implement a custom outbox in rtc fast memory?

euripedesrocha, 2024-06-07:
@Teesmo it will depend on your application and message types. 
From the mqtt client code it is possible and at first glance it could be a simple change from our current outbox implementation. You would need to change the allocations in [this function](https://github.com/espressif/esp-mqtt/blob/a0f4ce85de0e7cf05f04c2bdfedd65c1155aaf17/lib/mqtt_outbox.c#L42). You need to pay attention to other factors like the core that can access the memory you select and the available size to use.