espressif/esp-mqtt #279: Memory Leak when using esp_mqtt_client_publish

michaelboeding, 2024-05-18:
I seem to be running into a memory leak when using esp_mqtt_client_publish on esp idf v5.2.1. I was previously using esp idf v4.4. I'm also now using PSRAM so not sure if that has something to do with it also. The fix for me seems to be using qos 0 in my publish instead of qos 1. This didn't occur before I updated from v4.4 and updated all the apis to the latest version. I don't actually need qos 1 for my application so I just switched it to 0 for now but wanted to report it as a bug.

euripedesrocha, 2024-05-21:
Hi @michaelboeding could you point the scenario where you are getting the memory leak? Please share some reproducer and the configuration used for the MQTT client.

michaelboeding, 2024-08-12:
```c++

bool MQTTClient::setupMqttConnection(std::string certificate, std::string privateKey){

this->clientCertificate = certificate;
this->privateKey = privateKey;

ESP_ERROR_CHECK(nvs_flash_init());
ESP_ERROR_CHECK(esp_netif_init());

this->mqtt_cfg.broker.address.uri = "mqtts://<hidden>.iot.<region>.amazonaws.com:8883";
this->mqtt_cfg.broker.address.port = 8883;
this->mqtt_cfg.network.timeout_ms = 30000;
this->mqtt_cfg.network.reconnect_timeout_ms = 10000;
this->mqtt_cfg.session.keepalive = 1200;
this->mqtt_cfg.session.disable_keepalive = false;
this->mqtt_cfg.task.stack_size = 12500;
this->mqtt_cfg.buffer.size = 4096;
this->mqtt_cfg.broker.verification.certificate = AWS_ROOT_CA;
this->mqtt_cfg.credentials.authentication.certificate = this->clientCertificate.c_str();
this->mqtt_cfg.credentials.authentication.key = this->privateKey.c_str();

uint8_t lastWillPayload[1];
lastWillPayload[0] = 0;

std::string lastWillTopic = std::string(DEVICE_PREFIX) + "/" + get_device_mac_mqtt() + "/device/lastwill";
this->mqtt_cfg.session.last_will.topic = lastWillTopic.c_str();
this->mqtt_cfg.session.last_will.qos = 1;
this->mqtt_cfg.session.last_will.retain = false;
this->mqtt_cfg.session.last_will.msg = (const char*)lastWillPayload;
this->mqtt_cfg.session.last_will.msg_len = sizeof(lastWillPayload);

this->mqtt_cfg.session.protocol_ver = MQTT_PROTOCOL_V_3_1_1;
this->mqtt_cfg.credentials.client_id = get_device_mac_mqtt();

ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
client = esp_mqtt_client_init(&this->mqtt_cfg);

vTaskDelay(1000 / portTICK_PERIOD_MS);
if(client == NULL) {
    ESP_LOGE(TAG, "Failed to initialize MQTT client");
    return false;
}

esp_mqtt_client_register_event(client, MQTT_EVENT_ANY, mqtt_event_handler, this);
esp_mqtt_client_start(client);
return true;

}

```

Publish method 

int MQTTClient::publish(const std::string& topic, const uint8_t* data, size_t dataSize, int qos, bool retained) {
    if (!this->connected) {
        return -3;
    }

    checkHeapSize();
    getOutboxSize();

    int msg_id = esp_mqtt_client_publish(this->client, topic.c_str(), reinterpret_cast<const char*>(data), dataSize, qos, retained);

    checkHeapSize();
    getOutboxSize();

    if (msg_id == -1) {
        ESP_LOGE(TAG, "Error: MQTT publish failed");
        return msg_id;
    } else {
        ESP_LOGI(TAG, "MQTT publish succeeded, message ID: %d", msg_id);
        return msg_id;
    }
}

The above results in heap loss if the qos is set to 1

euripedesrocha, 2024-08-12:
@michaelboeding could you add the code for your `checkHeapSize`  and `getOutboxSize` functions?

In the `publish` we'll have a copy to the outbox and wait for the broker `PUBACK` control message to clear.  Could this be the cause?

closed 2024-10-02 by euripedesrocha