espressif/esp-mqtt #283: MQTT outbox dequeuing mechanism for esp_mqtt_client_enqueue

Teesmo, 2024-08-06:
How does esp_mqtt_client_enqueue dequeue? Is it a FIFO mechanism or a LIFO (or hybrid), I have noticed instances where the most recent data in the queue is published first, or is in conjuction with the (FIFO) queue(i.e temporarily interrupts FIFO dequeuing).

euripedesrocha, 2024-08-06:
Hi @Teesmo we always push items to the end. 
When trying to publish, the first valid occurrence is selected. 
For recently enqueued this translates to a FIFO, if we are trying to re transmit messages the one selected is the first that had expired the timeout for retransmission.

So it is always FIFO, but with some different criteria depending on scenario. 

Is this behavior affecting your system in some sense?

Teesmo, 2024-08-06:
I need a FIFO pipeline because I need data transferred in the order it was extracted from sensors (time-series is crucial here). I need this data not only for time-series but also in order processing, as I have external systems that interract with this system, and can alter the behaviour of this system.

euripedesrocha, 2024-08-06:
@Teesmo could you add more information like: QoS of messages, are you using publish or enqueue function?

Teesmo, 2024-08-06:
I am using the enqueue function, QoS 2

closed 2024-08-10 by Teesmo