espressif/esp-mqtt #287: Custom outbox not building

Teesmo, 2024-09-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I am implementing a custom outbox which is just a copy of mqtt_outbox.c with additional functions to persist the outbox to nvs. I copied the structure of the project level CMake file, and adapted it to include my C file (not the example C++ file). I removed redundant fucntions from my custom file (files in mqtt_outbox.c that I did not modify in my custom c file). I also renamed the functions that I modified, to avoid potential linker clash. However, when I build, I get this error:
`[1014/1016] Linking CXX executable mesh_local_control.elf
FAILED: mesh_local_control.elf 
: && /home/onepower/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32s2-elf-g++ -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S2=0 -Wl,--Map=/home/onepower/projects/esp/1meter_project/mesh_local_control/build/mesh_local_control.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s2.peripherals.ld -T esp32s2.rom.ld -T esp32s2.rom.api.ld -T esp32s2.rom.libgcc.ld -T esp32s2.rom.newlib-funcs.ld -T esp32s2.rom.newlib-data.ld -T esp32s2.rom.spiflash_legacy.ld -T memory.ld -T sections.ld CMakeFiles/mesh_local_control.elf.dir/project_elf_src_esp32s2.c.obj -o mesh_local_control.elf -L/home/onepower/esp/esp-idf/components/soc/esp32s2/ld   -L/home/onepower/esp/esp-idf/components/esp_rom/esp32s2/ld   -L/home/onepower/projects/esp/1meter_project/mesh_local_control/build/esp-idf/esp_system/ld   -L/home/onepower/esp/esp-idf/components/esp_phy/lib/esp32s2   -L/home/onepower/esp/esp-idf/components/esp_wifi/lib/esp32s2 esp-idf/xtensa/libxtensa.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_dac/libesp_driver_dac.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_psram/libesp_psram.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__esp_modem/libespressif__esp_modem.a  esp-idf/espressif__usb_host_cdc_acm/libespressif__usb_host_cdc_acm.a  esp-idf/espressif__esp_modem_usb_dte/libespressif__esp_modem_usb_dte.a  esp-idf/espressif__tinyusb/libespressif__tinyusb.a  esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a  esp-idf/espressif__iot_bridge/libespressif__iot_bridge.a  esp-idf/espressif__mesh_lite/libespressif__mesh_lite.a  esp-idf/joltwallet__littlefs/libjoltwallet__littlefs.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/espcoredump/libespcoredump.a  -u esp_system_include_coredump_init  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/espressif__mesh_lite/libespressif__mesh_lite.a  /home/onepower/projects/esp/1meter_project/mesh_local_control/managed_components/espressif__mesh_lite/lib/libesp_mesh_lite_esp32s2.a  esp-idf/espressif__mesh_lite/libespressif__mesh_lite.a  /home/onepower/projects/esp/1meter_project/mesh_local_control/managed_components/espressif__mesh_lite/lib/libesp_mesh_lite_esp32s2.a  esp-idf/json/libjson.a  esp-idf/espressif__iot_bridge/libespressif__iot_bridge.a  esp-idf/espressif__esp_modem_usb_dte/libespressif__esp_modem_usb_dte.a  esp-idf/espressif__esp_modem/libespressif__esp_modem.a  esp-idf/espressif__usb_host_cdc_acm/libespressif__usb_host_cdc_acm.a  esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/usb/libusb.a  esp-idf/espressif__tinyusb/libespressif__tinyusb.a  esp-idf/joltwallet__littlefs/libjoltwallet__littlefs.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/console/libconsole.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_dac/libesp_driver_dac.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf [... cut ...]
/home/onepower/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:2267:(.text.esp_mqtt_client_get_outbox_size+0x16): undefined reference to `outbox_get_size'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
`

To sanity check, I copied the custom mqtt example to a separate folder (out of idf), and it built perfectly. My custom outbox implementation (almost all the files) is technically the same as the example C++ custom example, and I was expecting that I would not encounter any errors since I have copied evereything 'almost' to the tee.

euripedesrocha, 2024-09-24:
Hi @Teesmo If I understood correctly you only implemented in your custom outbox the functions you needed to be customized. 
To have a custom outbox, you need to implement all functions.

Teesmo, 2024-09-24:
I tried copying and pasting mqtt_outbox.c, and added a few other functions, and that didn't work as well (in this scenario, I kept everything from mqtt_outbox.c). This error specifically says all those functions in mqtt_outbox.c are not defined, or something to that effect.

euripedesrocha, 2024-09-24:
@Teesmo once you enable the custom outbox, all implementation for the outbox will be removed. You then need to add your custom functions for the build.

You can see [at this line](https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/examples/protocols/mqtt/custom_outbox/CMakeLists.txt#L9) how to add your sources. 

If you can share the cmake file and the content of your customized implementation, I can give you other suggestions to fix.

Teesmo, 2024-09-24:
Here is my CMake

`cmake_minimum_required(VERSION 3.16)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(mesh_local_control)
idf_component_get_property(mqtt mqtt COMPONENT_LIB)
target_sources(${mqtt} PRIVATE ${CMAKE_CURRENT_SOURCE_DIR}/main/custom_outbox.c)`

and here is my custom outbox
```
#include <stdint.h>
#include <stdlib.h>
#include <string.h>
#include "mqtt_config.h"
#include "sys/queue.h"
#include "esp_heap_caps.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "nvs.h"
#ifndef CONFIG_MQTT_CUSTOM_OUTBOX

static const char *TAG = "outbox";

typedef struct outbox_item {
    char *buffer;
    int len;
    int msg_id;
    int msg_type;
    int msg_qos;
    outbox_tick_t tick;
    pending_state_t pending;
    STAILQ_ENTRY(outbox_item) next;
} outbox_item_t;

STAILQ_HEAD(outbox_list_t, outbox_item);

struct outbox_t {
    _Atomic uint64_t size;
    struct outbox_list_t *list;
    nvs_handle_t nvs_handle;
};

esp_err_t save_outbox_to_nvs(outbox_handle_t outbox)
{
    esp_err_t err;
    
    // Save the size
    err = nvs_set_u64(outbox->nvs_handle, "size", outbox->size);
    if (err != ESP_OK) return err;

    // Save the number of items
    uint32_t item_count = 0;
    outbox_item_handle_t item;
    STAILQ_FOREACH(item, outbox->list, next) {
        item_count++;
    }
    err = nvs_set_u32(outbox->nvs_handle, "item_count", item_count);
    if (err != ESP_OK) return err;

    // Save each item
    uint32_t i = 0;
    STAILQ_FOREACH(item, outbox->list, next) {
        char key[32];
        snprintf(key, sizeof(key), "item_%ld", i);
        
        err = nvs_set_blob(outbox->nvs_handle, key, item, sizeof(outbox_item_t));
        if (err != ESP_OK) return err;

        snprintf(key, sizeof(key), "buffer_%ld", i);
        err = nvs_set_blob(outbox->nvs_handle, key, item->buffer, item->len);
[... 254 lines cut ...]
{
    return outbox->size;
}

void outbox_delete_all_items(outbox_handle_t outbox)
{
    outbox_item_handle_t item, tmp;
    STAILQ_FOREACH_SAFE(item, outbox->list, next, tmp) {
        STAILQ_REMOVE(outbox->list, item, outbox_item, next);
        outbox->size -= item->len;
        free(item->buffer);
        free(item);
    }
    save_outbox_to_nvs(outbox);
}
void outbox_destroy(outbox_handle_t outbox)
{
    outbox_delete_all_items(outbox);
    nvs_close(outbox->nvs_handle);
    free(outbox->list);
    free(outbox);
}

#endif /* CONFIG_MQTT_CUSTOM_OUTBOX */
```

Teesmo, 2024-09-24:
I figured that my issues were created by this line (and its closing counterpart)
`#ifndef CONFIG_MQTT_CUSTOM_OUTBOX`
My issue seems to have been solved since I deleted it.

euripedesrocha, 2024-09-25:
@Teesmo yes. We use the mechanism to prevent the outbox functions to b included in the build if the custom outbox is enabled. I'm assuming that I can close this issue, feel free to reopen if needed.

closed 2024-09-25 by euripedesrocha