espressif/esp-mqtt #291: In Tag version ESP-MQTT_FOR_IDF_3.1,I encounter a problem.This seems to be an issue with buffer overflow

cortex-m4, 2024-11-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

When I complier ESP8266 RTOS project ↓

esp-mqtt/lib/transport_ws.c:100:53: error: '258EAFA5-E914-47DA-95CA-C5AB...' directive writing 36 bytes into a region of size between 33 and 64 [-Werror=format-overflow=]
     int key_len = sprintf((char*)client_key_b64, "%s258EAFA5-E914-47DA-95CA-C5AB0DC85B11", (char*)client_key);
esp-mqtt/lib/transport_ws.c:100:19: note: 'sprintf' output between 37 and 68 bytes into a destination of size 64
     int key_len = sprintf((char*)client_key_b64, "%s258EAFA5-E914-47DA-95CA-C5AB0DC85B11", (char*)client_key);
                   
cc1: some warnings being treated as errors

cortex-m4, 2024-11-13:
Use version ESP-MQTT_FOR_IDF_4.x instead of ESP-MQTT_FOR_IDF_3.1 can solve this question.

euripedesrocha, 2024-11-15:
@cortex-m4 I assume that you solved your issue. Feel free to reopen If I misunderstood.

closed 2024-11-15 by euripedesrocha