espressif/esp-mqtt #292: Memory allocation

allacmc, 2024-11-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I have a question or a suggestion:

When we start MQTT, does it take a copy of the digital certificate and make its own memory allocation or does it use the indicated pointer to access the digital certificate?

I noticed in some tests I carried out that if I mark the digital certificate pointer as free(), MQTT does not work correctly.

Therefore, I need to keep the digital certificate data pointer in memory.

As can be seen below: a part of the code with this context:

`
void f_startMQTT_Out(){
    xTaskCreate(f_setupMQTT_Out, "f_setupMQTT_Out", 2400, NULL, tskIDLE_PRIORITY, NULL);
    ESP_LOGI(TAG, "Startado MQTT_Out");
}

void f_setupMQTT_Out(void *pvParameters){
        if (xEventGroupMQTT_Out == NULL){
            xEventGroupMQTT_Out = xEventGroupCreate();    
        }
        Dados_Mqtt_t DadosMqttout;
        if(f_DadosMqtt(&DadosMqttout, mqtt_OUT) == ESP_OK){
                char * mqtt_id_Out  = f_mqttId_Out();
                char * topico = f_topico(zMQTT_sufixo);
                char * lastmsg = f_prepJsonConfig(mqtt_OFF);
                size_t len_lastmsg = strlen(lastmsg);
                bool is_secure = strstr(DadosMqttout.mqtt_server, "mqtts://") == DadosMqttout.mqtt_server;
                char *cert_data_out = NULL; // Certificado inicializado como NULL
                if (is_secure) {
                    cert_data_out = f_lerCertificado(DadosMqttout.mqtt_cert);
                    if (cert_data_out == NULL) {
                        ESP_LOGE("MQTT", "Falha ao carregar o certificado: %s", DadosMqttout.mqtt_cert);
                    }
                }
                esp_mqtt_client_config_t mqtt_cfg_out = {
                    .broker.address.uri = DadosMqttout.mqtt_server,
                    .broker.address.port = DadosMqttout.mqtt_port,
                    .credentials.username = DadosMqttout.mqtt_user,
                    .credentials.authentication.password = DadosMqttout.mqtt_pass,
                    .credentials.client_id = mqtt_id_Out,
                    .session.last_will.msg = lastmsg,
                    .session.last_will.msg_len = len_lastmsg,
                    .session.last_will.topic = topico,
                    .session.last_will.retain = true,
                    .session.last_will.qos = 1,
                    .session.keepalive = 120, // DadosMqttout.mqtt_keepalive
                    .task.stack_size = 3200,
                    //.task.priority = tskIDLE_PRIORITY + 1
                };
                if (is_secure && cert_data_out != NULL) {
                    mqtt_cfg_out.broker.verification.certificate = cert_data_out;
                    mqtt_cfg_out.broker.verification.certificate_len = strlen(cert_data_out) + 1;
                }
                EventBits_t bits = xEventGroupWaitBits(xEventGroupWifi, WIFI_BIT_0, pdFALSE, pdFALSE, portMAX_DELAY);
                if((bits & WIFI_BIT_0)){
                        cMqttOut = esp_mqtt_client_init(&mqtt_cfg_out);
                        esp_mqtt_client_register_event(cMqttOut, ESP_EVENT_ANY_ID, mqtt_event_handler_out, NULL);
                        esp_mqtt_client_start(cMqttOut);
                }
                safe_free(mqtt_id_Out);
                safe_free(topico);
                safe_free(lastmsg);
                //safe_free(cert_data_out); ??Se eu marcar esse ponteiro como free após iniciar o MQTT ele não funciona perfeitamente
        }
        f_memoria("f_setupMQTT_OUT");
        vTaskDelete(NULL);
}
`

euripedesrocha, 2024-11-18:
Hi @allacmc thanks for reporting.
As the [documentation states](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mqtt.html#_CPPv4N24esp_mqtt_client_config_t8broker_t14verification_t11certificateE) the lifetime of the certificate data is for the user to control. I can see that from the way it is written it might cause some confusion. It needs to be valid through the life of the mqtt client, and it's up to you to clean it after. 

I'll try to improve the documentation in that sense.

allacmc, 2024-11-19:
Thank you for answering my question, I was able to find it in the documentation.

It doesn't actually make a copy of the digital certificate, so I must keep the pointer with the allocated certificate at all times.

As always you are the best ESP32 EURIPEDES connectivity dev.

Thank you for working on the ESP-IDF project.

I love this framework

closed 2024-11-19 by allacmc