espressif/esp-mqtt #294: mqtt does not url decode username or password (IDFGH-14444)

0xFEEDC0DE64, 2025-01-16:
### Is your feature request related to a problem?

yes, our customers have mqtt servers with usernames and passwords containing spaces, @ and other special characters.

Telling them to correctly urlencode the parameters didnt help.

We only expose one single mqtt connection url and not the individual username and password fields.

### Describe the solution you'd like.

The mqtt library should urldecode automatically if the username and passwords are taken from the url

euripedesrocha, 2025-01-16:
@0xFEEDC0DE64 I moved the issue. 

Could you add more details on use case, failures that are preventing, if possible some test cases and the broker version so if we decide to implement the feature we can test it?

erxbout, 2025-01-26:
I previously linked our issue a customer opened

Here is the link to the problem:
https://github.com/goecharger/go-eCharger-API-v2/issues/258#issuecomment-2595704954

theNorr, 2025-11-26:
You can use me for troubleshooting. I have the Swedish letters ÅÄÖ in my password and that blocks me from saving the URL.

What do you need from me, to be able to troubleshoot this further?