espressif/esp-mqtt #303: [Documentation] How to limit output queue size?

doragasu, 2025-07-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Currently I am using the esp-mqtt-client with the `esp_mqtt_client_enqueue()` function. Everything works great so far, but I would like to know how the output queue mechanism works. Browsing the documentation, I see there are two parameters that might be related to the output queue:

* `out_size` inside the `struct buffer_t`. According to documentation, this is the size of the MQTT output buffer.
* `limit` inside the `outbox_config_t`: According to documentation, this is the "size limit for the outbox in bytes".

To me it's not clear how these parameters work exactly. It seems there is no limit in the number of messages enqueued, only in the buffer length. But how can I predict how many messages will I be able to enqueue? Because I suppose in addition to the data to send (topic and payload), there must be some overhead added to this buffer. Also, what are the defaults for these parameters?

doragasu, 2025-07-08:
@someburner Thanks for the info. I don't really need to limit the number of inflight messages. I just want to know how to configure esp-mqtt-client in order to allow a minimum amount of messages to be enqueued (my problem was just with the documentation). Anyway I have read the issue you linked, and learned about the blocking nature of `esp_mqtt_client_enqueue()`, I didn't know that and it can cause trouble, will have to look into it.

@euripedesrocha Oh, I missed some checks but now I see outbox_limit is checked against outbox size plus data length. I don't know why would you want to limit the size, but now I understand how it works, thanks!

euripedesrocha, 2025-07-08:
@doragasu the feature was introduced to allow users to control the memory usage, and memory is a limited resource in embedded environment. We decided to keep only memory, instead of controlling the number of messages because depends on the application. There might be scenarios where the user application have different types of message payloads, so total memory was the number to constraint. 

As mentioned you can customize the outbox, the process involves defining that you are using a custom outbox and applying the changes you want. For this specific case, it shouldn't be too complicated to store the number of messages. 

You can also adopt MQTT5 that has it integrated in the protocol if possible.

doragasu, 2025-07-09:
@euripedesrocha: But doesn't the `out_size` parameter in `struct buffer_t` already limit the memory footprint? I thought this `limit` was an additional restriction when pushing data into the output buffer, but the allocated buffer length is specified in `out_size` if I didn't misunderstand the code.

doragasu, 2025-07-10:
Thanks for the clarification. I already saw the mqtt_outbox.c does indeed allocate memory for each message.

dieter264, 2025-11-11:
Small correction to my previous comment: I’ll use the `tcp` example instead of the `custom_outbox` example to test the default outbox behavior.

doragasu, 2025-11-14:
Awesome news, I could not work in this, so the PR is welcome!

doragasu, 2025-11-25:
This looks awesome!