espressif/esp-mqtt #306: Are the all features of MQTT v5 already implemented in this SDK?

shijunjun521, 2025-07-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Does this SDK fully support all MQTT v5 features? As we know, Paho MQTT Embedded-C only partially implements MQTT v5, so we prefer an SDK with complete feature support.

### Use cases

complete feature support

euripedesrocha, 2025-07-25:
Hi @shijunjun521 As far as I know we have one control message to be implemented, AUTH. 

Do you have any specific feature of MQTT v5 that you are looking for?

shijunjun521, 2025-07-25:
@euripedesrocha Thanks for the reply. Basicly, I would like common features of  V5 to meet costomers' requirements.  And for some sepecilal requirements ,the full-implement SDK is perferred. So does the current SDK meet most MQTT application scenarios?

euripedesrocha, 2025-07-30:
@shijunjun521 looking at the list provided, It looks like we have everything but the enhanced authentication which is the AUTH control message I stated before. Have you tried our mqtt5 example?

shijunjun521, 2025-08-01:
@euripedesrocha Got it. Yes , I have tried the examples, it works well.