espressif/esp-mqtt #307: Unable to query client to find current status

hachi, 2025-07-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The client API provides no way to query if it is currently connected, reconnecting, disconnected, or entirely stopped.

I think there's a usability bug because if we issue a set_uri or set_config call we need to know the state of the client to decide whether to disconnect, start, or maybe something else.

### Actual behavior (suspected bug)

client init with config that has invalid uri will result init then start, start will end and task will shutdown

later come back to client and issue set_uri or set_config followed by disconnect to trigger reconnect, no action taken because task is stopped

This is more apparent if you try to create a watchdog task to check your mqtt client is running correctly, no way to query client and ask if running/connected

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. Write application
2. Consult docs
3. Question state machine
4. No state machine query functions

### Project release version

latest esp-idf release, and master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Windows Vista

### Shell

ZSH

euripedesrocha, 2025-07-30:
@hachi thanks for the suggestion. 
I have some points:
- start a client with invalid uri should return NULL, hence an error, do you have a case where this wasn't true? I believe that here is the core problem. 
If you set config or uri the flow, which I agree could be improved, will depend on the configuration you have for the client reconnection. I would like to look at how you are using the esp_mqtt client. If you could provide a code snippet with your config struct and how you are managing the start, it would be very helpful. 

I'm not against adding the new api, but we may need a broader change on this workflow to achieve a better result.

hachi, 2025-07-30:
Yes actually, it's happily creating the client without a broker to connect to. Would this be the subtle difference between an empty string and a NULL?

hachi, 2025-07-30:
Yeah, I'm like 90% sure that's what gets me into that state, I pass the value I have in NVS in as the URL, and I default to NULL if NVS is empty.

I don't think you should guard against this either, because a valid looking URI can also result in no connection as well. Besides the behavior change that this would cause for people.

I think most people using this assume the client init would only fail if some resource failure occurs, the examples do not demonstrate proper guarding.

euripedesrocha, 2025-07-30:
@hachi thanks for clarification, this looks like a bug. It was potentially introduced when we moved the transport initialization to allow custom transport usage. 

A NULL url is only valid scenario if the broker is set using the other parameters. If you miss all parameters, it should be a failure on init. I will fix this bug. 

I'm still interested in your reconfigure workflow though, if there is an improvement to be made I would like to handle that as well.

euripedesrocha, 2025-07-31:
@hachi thanks again for clarifying. 
I will check the behavior with empty string. I believe that you shouldn't be able to init the mqtt client with a broken broker address configuration. 

For the remaining issues, we'll address them later, as there are some breaking changes.

hachi, 2025-08-15:
I'm still looking for traction on this topic. I'm not sure how the expected use is for changing configs and reconnecting.

`esp_mqtt_set_config` does not trigger a reconnection if the broker changes
`esp_mqtt_client_reconnect` doesn't reconnect unless you are already in the disconnected state
`esp_mqtt_client_connect` is not a thing
`esp_mqtt_client_disconnect` queues up a disconnect, followed by the reconnect semantics...

should calling like this

```
_set_config
_disconnect
_reconnect
```

all without yielding trigger an immediate switch to the new broker? (I know there will be protocol/task delays, but I want this to happen without the reconnect delay)

hachi, 2025-09-03:
@euripedesrocha Did you check the behavior of empty string yet?

euripedesrocha, 2025-09-03:
@hachi yes, I have an internal fix that is waiting for some CI related fixes for testing. I'll take a different approach to it, to make it available on master sooner.

hachi, 2025-09-10:
Only because I have no config yet, so my mqtt broker string is unset and through code ends up just passing in empty string.

I don't need the function to work, but you should probably not break the interface's behavior without a major version change.

It would be difficult to decide if a broker URI is valid without trying to start the client, and the implication in the API is that you can set up a config and then update the URI before starting it up... if you can reject URI before starting up how will you decide what is valid vs not valid?

Again though, this ticket was intended to ask for a `esp_mqtt_client_get_status(&status)` sort of call, so you can query for 'init' 'connecting' 'connected' 'reconnecting' 'disconnected' 'stopped' status of the client.

hachi, 2025-09-10:
Why does the client try to set up transport during config parsing? Shouldn't that be done during `_start` ? If a person updates the URI after config parse before start, then the transport may change and you would have to set up again. No reason to set up transport until you need it.

nebkat, 2026-08-15:
> [@nebkat](https://github.com/nebkat) I would like to clarify some aspects. What is your end goal on the init state? The only way to call the `get_state` function and the client responding is in that state is on the before_connect event, due to the lock in the function.

`MQTT_CLIENT_STATE_NOT_STARTED` is returned in two places:
1) `if (client->task_handle == NULL)`
2) `switch (client->state) { case MQTT_STATE_INIT:`

If the task has not been started one can simply call `esp_mqtt_client_destroy` and be certain that it will return immediately. If the task _has_ been started and is sitting in `MQTT_STATE_INIT` it is necessary to call `esp_mqtt_client_stop` first, which could get stuck if a connection has started since your `get_state` call. So we should at least differentiate between them.

_Of course it is possible to track whether one has called `client_start` in the first place, but we could get that information for free from the client._

I settled on the following fix: https://github.com/nebkat/esp-mqtt/commit/01a1414947ea6361a6361dacc74b758223c4305e, though since then I have also solved the nondeterministic stop by introducing forced stops: https://github.com/nebkat/esp-mqtt/commit/3a46a1b6ce03e5c2da41c972f1ff8438750fe6ab.

closed 2026-08-18 by david-cermak