espressif/esp-mqtt #312: mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ack

hachi, 2025-08-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

No warning

### Actual behavior (suspected bug)

After a number of broker reconnects I receive that warning/error.

### Error logs or terminal output

```shell
E (00:33:14.513) mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ack                                                                                                             I (00:33:14.516) mqtt_client: MQTT5 publish check fail                                                                                                      
E (00:33:14.613) mqtt5_client: Client send more than 20 QoS1 and QoS2 PUBLISH packet without no ack
I (00:33:14.615) mqtt_client: MQTT5 publish check fail
```

### Steps to reproduce the behavior

Break connection to the broker a number of times, it just shows up eventually.

### Project release version

ESP-IDF 5.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

This is not a build time problem

### Shell

ZSH

### Additional context

I am aware of #261 but I believe this should have been fixed long before ESP-IDF 5.5

I do not know how to debug this further, but am curious if you think this bug is fixed in ESP-IDF 5.5

#261 does not list a commit where the fix was applied

hachi, 2025-08-20:
In my testing I am finding that esp_mqtt_client_publish is not blocking until PUBACK... is this intended behavior?

hachi, 2025-08-21:
```
I (00:55:32.120) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:32.772) mqtt5_client: PUBLISHED 0 WITHOUT ACK 
I (00:55:32.926) mqtt5_client: PUBLISHED 1 WITHOUT ACK 
I (00:55:33.079) mqtt5_client: PUBLISHED 2 WITHOUT ACK 
I (00:55:33.179) mqtt5_client: PUBLISHED 3 WITHOUT ACK 
I (00:55:33.279) mqtt5_client: PUBLISHED 4 WITHOUT ACK 
I (00:55:33.379) mqtt5_client: PUBLISHED 5 WITHOUT ACK 
I (00:55:33.478) mqtt5_client: PUBLISHED 6 WITHOUT ACK 
I (00:55:33.540) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:33.553) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:33.578) mqtt5_client: PUBLISHED 5 WITHOUT ACK 
I (00:55:33.678) mqtt5_client: PUBLISHED 6 WITHOUT ACK 
I (00:55:34.011) mqtt5_client: PUBLISHED 7 WITHOUT ACK 
I (00:55:34.060) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.065) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.070) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.074) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.082) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.089) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.137) mqtt5_client: PUBLISHED 2 WITHOUT ACK 
I (00:55:34.165) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.171) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:34.268) mqtt5_client: MQTT_MSG_TYPE_PUBACK return code is 0
I (00:55:35.313) mqtt5_client: PUBLISHED 0 WITHOUT ACK 
```

I added a debug log line of the current count any time it is evaluated. It looks like during times of high latency I can end up with multiple messages in flight before the PUBACK is returned for them. As a result I can sometimes overrun the limit set on the server.

The documentation states: `QoS 1 and 2 messages that may need retransmission are always enqueued, but first transmission try occurs immediately if esp_mqtt_client_publish is used.`

But then also states: `This API might block for several seconds, either due to network timeout (10s) or if publishing payloads longer than internal buffer (due to message fragmentation)`

This seems to imply that unless we time out (and need to retry) it should block until PUBACK is received, but isn't doing that.

The flow control isn't controlling the flow, so you can run above the high watermark.

Additionally though, the safety check then drops the message rather than enqueueing it for delivery once the flow control frees up.

I am not reaching the timeout, and I am only publishing from a single task/thread.

hachi, 2025-09-03:
@euripedesrocha @bogdankolendovskyy is this a bug or not?

euripedesrocha, 2025-09-30:
@hachi, thanks for reporting. 
The publish function will block only until the message is sent. We don't block until PUBACK. Similar behavior occurs when you are using QoS2 messages. 
I might be biased here, but that documentation never implied the blocking until completing the full message processing, we'll clarify it thanks for pointing out.

For the message dropping behavior, we can improve the outbox to track message state+number of messages sent. I'll try to address it in next release. 

Thanks again for your contributions in pointing these improvements, very much appreciated.

euripedesrocha, 2025-10-02:
Ok, I need to double-check the log messages as there are no PUBACK involved in QoS 2, anyway the behavior for enqueueing should be the same. 

I don't intend to create a double-queue system, but I understand your point here. Our internal check function is restrictive in this sense. I was a bit off from your usage of the word drop, I was understanding that you were enqueueing and message being dropped after that, sorry for misunderstanding you. 

This was a decision to simplify outbox management by preventing messages that we wouldn't be able to publish to enter. 
We'll put some work on this to improve the behavior.

hachi, 2026-01-28:
@euripedesrocha any progress on this?

hachi, 2026-08-24:
I'm not sure but I think this was showing up as a bug in mosquitto as well, I have since started running mosquitto 2.1.2 and my older clients are triggering a bug I've reported over on their issue tracker.

The good news, I think... is that this fix may have fixed that bug for newer clients.

This doesn't help people running IDF < 6... unless you manually update to newer ESP-MQTT.

I think the mosquitto project may be dead. There haven't been many updates to it, no releases, and no responses to tickets about this and many other reported bugs to it for a while.

So the bottom line I think is that the combination of: ESP-IDF < 6 with default ESP-MQTT, MQTT protocol v5, and Mosquitto 2.1.2 (the current stable release) can lead to network latency being translated directly into the broker forcibly disconnecting clients.

hachi, 2026-08-24:
Oh I forgot to link the issue I reported over there: https://github.com/eclipse-mosquitto/mosquitto/issues/3699

euripedesrocha, 2026-08-25:
Hi @hachi thanks for your inputs. 
For idf from 5.3 to 5.5 it is possible to use esp-mqtt as a managed component. For some things I will create backports but I believe the managed component path is an easier workaround. 

Regarding the disconnect path, that might be on esp-mqtt. 

Mqtt 3.11 is more flexible in terms of retransmission while Mqtt 5.0 is more strict on when it is allowed to re-transmit. Mosquitto might be inforcing that. In original implementation esp-mqtt took the path of prioritizing the message delivery and teh re-transmission happens without disconnection. I identified the slightly different wording on the spec while working on the conformance test that I've been gradually introducing.